mikuso/ocpp-rpc #6: Add support for JWS signed messages

mikuso, 2022-04-27:
Section 7.1 of the OCPP2.0.1J specification describes a "signed message format" which extends the RPC framework.

Support for this could be added with an option to the `client.call()` method.

e.g.
```js
await client.call('MeterValues', {/** params **/}, { sign: true });`
```

ArvindSteamA, 2023-04-26:
Hi @mikuso , this enhancement is still to be added right?

ArvindKarthi, 2023-04-26:
Hi @mikuso , this enhancement is yet to be done?

mikuso, 2023-04-26:
Yes. I feel like the spec isn't clear enough in how this should be implemented that I don't have confidence in my ability to implement it correctly. The biggest hurdle with this one is lack of a concrete example in the spec. I have asked the OCA for an example but didn't get any response.

I suppose my biggest worry is that I would accidentally release an implementation which wasn't compatible with other implementations.

Do you know of any other implementations that support this feature? It might be useful as a basis for me to test against.

ArvindKarthi, 2023-04-26:
Yes, your point is understandable. Currently I have not seen other implementations with this feature but let me check, it'll be a good security feature.