mikuso/ocpp-rpc #13: About strict mode and ajv

alex1290, 2022-05-31:
nodejs version: v18.2.0

```
        const server = new RPCServer({
            protocols: ['ocpp2.0.1'],
            strictMode: true,
        });
       
       // after client connected
       client.call('RequestStartTransaction', requestStartTransactionPayload);
```
After I lauch server, I tried to call "RequestStartTransaction" request.
And I got Error: ```strict mode: "additionalItems" is ignored when "items" is not an array of schemas.```
So I delete the all of "additionalItems" attirbutes in "ocpp2_0_1.json" and it works.

I think it is because of this: https://ajv.js.org/strict-mode.html#ignored-additionalitems-keyword

But I tried to not use strictMode when implement server.
If I tried to call the client, the client will be disconnected.

mikuso, 2022-05-31:
Thanks for the report.

Can you please provide the content of the `requestStartTransactionPayload`?

alex1290, 2022-06-01:
```
const requestStartTransactionPayload = {
            remoteStartId: new Date().getTime(),
            idToken: {
                idToken: user_uid, // string
                type: 'Central'
            }
        };
```
Do I miss something required?

mikuso, 2022-06-01:
Thanks for the report.  You are correct - it seems to be a problem with Ajv's strict validation of the schema.

I've decided to disable the strictSchema validation option in the Ajv constructor.  This should add some flexibility without harming interoperability.

I may expose the Ajv constructor options in a future update if there is a request for it.

This is now fixed in v1.5.1 4a7ed6e5b48e2a81b5df5f9f97b987d23fce11af

closed 2022-06-01 by mikuso