mikuso/ocpp-rpc #14: How to get response from server to client? When any method has been called from client?

kvext, 2022-06-02:
Hi @mikuso ,

I used the package ocpp-rpc and have query regarding receiving response from server to client. so can you please provide your suggestion on it?

I have created two separate folder one is for server and another is for client and install the ocpp-rpc library on both app.

On the server end, I have already handled the method as below:
```
server.on('client', async (client) => {
       await client.handle('MeterValues', ({params}) => {
        console.log(`Server got MeterValues from ${client.identity}:`, params);
        return {};
    });
});
```
Here, If I will return JSON object with any properties, It will not allow me and generate error on client end and aborted it.
event if I will return like `return {  status: "Accepted" }` on server end then it will show error on client as below:
**So, here how can we get response from server to client end?** **and Make sure that the method has been called successfully.**

```
\node_modules\ocpp-rpc\lib\util.js:44
    const err = new E(message ?? '');
                ^

RPCPropertyConstraintViolationError: data must NOT have additional properties
    at createRPCError (\client_app\node_modules\ocpp-rpc\lib\util.js:44:17)
    at Validator.validate (\client_app\node_modules\ocpp-rpc\lib\validator.js:47:19)
    at RPCClient._onCallResult (\client_app\node_modules\ocpp-rpc\lib\client.js:822:31)
    at RPCClient._onMessage (\client_app\node_modules\ocpp-rpc\lib\client.js:644:26)
    at WebSocket.<anonymous> (\client_app\node_modules\ocpp-rpc\lib\client.js:369:42)
    at WebSocket.emit (node:events:527:28)
    at Receiver.receiverOnMessage (\client_app\node_modules\ws\lib\websocket.js:1169:20)
    at Receiver.emit (node:events:527:28)
    at Receiver.dataMessage (\client_app\node_modules\ws\lib\receiver.js:528:14)
    at Receiver.getData (\client_app\node_modules\ws\lib\receiver.js:446:17) {
  rpcErrorMessage: 'Payload is syntactically correct but at least one field contains an invalid value',
  rpcErrorCode: 'PropertyConstraintViolation',
  details: {
    errors: [
      {
        instancePath: '',
        schemaPath: '#/additionalProperties',
        keyword: 'additionalProperties',
        params: { additionalProperty: 'status' },
        message: 'must NOT have additional properties'
      }
    ],
    data: { status: 'Accepted' }
  }
}
```
Calling the method from Client as below:

```
const cli = new RPCClient({
    endpoint: "ws://localhost:3000",
    identity: "EXAMPLE123",
    protocols: ["ocpp1.6"],
    strictMode: true,
});

cli.call("MeterValues", {
    connectorId: 0,
    transactionId: 12345,
    meterValue: [
        {
            sampledValue: [
                {
                    unit: "Celsius",
                    context: "Sample.Clock",
                    measurand: "Temperature",
                    location: "Body",
                    value: "33.8,18.6",
                },
            ],
            timestamp: new Date().toISOString(),
        },
    ],
});
```

Thanks,

mikuso, 2022-06-02:
Hi,

`cli.call()` returns a Promise which will either resolve with the value of the response from the other side, or it will reject with an RPCError of some sort.  If the client uses `strictMode: true`, then it will also reject with an RPCError if the response is invalid.

If the Promise from `cli.call()` resolves to a value, then you can be sure that the server has responded correctly to the call.

I have expanded on your example to demonstrate:

server.js
```js
const {RPCServer} = require('ocpp-rpc');

const server = new RPCServer({
    protocols: ['ocpp1.6'],
    strictMode: true,
});

server.on('client', async (client) => {
    await client.handle('MeterValues', ({params}) => {
        console.log(`Server got MeterValues from ${client.identity}:`, params);
        return {};
    });
});

server.listen(3000);
```

client.js
```js
const {RPCClient} = require('ocpp-rpc');

async function main() {
    const cli = new RPCClient({
        endpoint: "ws://localhost:3000",
        identity: "EXAMPLE123",
        protocols: ["ocpp1.6"],
        strictMode: true,
    });

    await cli.connect();
    
    try {
        const response = await cli.call("MeterValues", {
            connectorId: 0,
            transactionId: 12345,
            meterValue: [
                {
                    sampledValue: [
                        {
                            unit: "Celsius",
                            context: "Sample.Clock",
                            measurand: "Temperature",
                            location: "Body",
                            value: "33.8,18.6",
                        },
                    ],
                    timestamp: new Date().toISOString(),
                },
            ],
        });

        console.log('Got reply:', response); // Got reply: {}
    } catch (err) {
        // MeterValues call failed

        console.error("MeterValues call failed because:", err.message);
        console.error("RPC Error code:", err.rpcErrorCode);
        console.error("Error details:", err.details);
    }
}

main().catch(console.error);
```

Does this help?

mikuso, 2022-06-02:
For completeness, if you want to test sending an error from the server:

```js
server.on('client', async (client) => {
    await client.handle('MeterValues', ({params}) => {
        throw createRPCError("InternalError", "This is an example error");
    });
});
```

closed 2022-06-02 by mikuso