mikuso/ocpp-rpc #21: Waiting for server connection

LMJayasundara, 2022-11-03:
I am try to implemt waiting for server connection function using this lib. I can not handdle it. Can you give any idea for it.
Thanks

LMJayasundara, 2022-11-03:
```
async connect(){
        this._ws = new WebSocket(this._URL, {
            perMessageDeflate: false,
            headers: {
                Authorization: Buffer.from(this._username).toString('base64'),
            },
        });

        try {
            await this._beginConnect();
        } catch (error) {
            console.log(error.message);
        }
    }

    async _beginConnect() {
        try {
            await new Promise((resolve, reject) => {    
                this._ws.on('open', () => {
                    console.log('open');
                    clearInterval(this._reconn);
                    resolve();
                });

                this._ws.on('error', error => {
                    reject(error);
                });
                
                this._ws.on('close', () => {
                    this._tryReconnect();
                    reject();
                });
            });

            this._attachWebsocket(this._ws);

        } catch (error) {
            console.log(error.message);
            this._ws.terminate();
            this.connect();
        }

        // console.log("Begin Connectx: ", this._ws.readyState);
        // this.emit('state', {state: this._ws.readyState});
    }

    async _tryReconnect() {
        this._reconn = setTimeout(async() => {
            this._ws.terminate();
            await this.connect();
        }, 5000);
    }

    async _attachWebsocket(ws) {
        process.nextTick(() => {
            ws.on('message', msg => this._onMessage(msg));
        });
    }
```

mikuso, 2022-11-20:
Sorry I don't understand your question. Can you elaborate? It doesn't look like your code example uses this library.

mikuso, 2023-01-10:
Closing this issue as it seems to have gone stale.

closed 2023-01-10 by mikuso