mikuso/ocpp-rpc #22: Example for remoteStopTransaction / remoteStartTransaction

zelix25, 2022-11-23:
Hello, could you explain to me which function I must use to carry out the remoteStopTransaction / remoteStartTransaction  commands via the central server?
I can't figure out how it works...

Thanks

mikuso, 2022-11-23:
Hi,

It sounds like you're talking specifically about OCPP version 1.6 ?

RemoteStartTransaction and RemoteStopTransaction are calls sent from the server to the client.

Here's an example of how to set up an OCPP1.6 server which sends a RemoteStartTransaction request whenever a client connects:
```js
const {RPCServer} = require('ocpp-rpc');

const server = new RPCServer({
    protocols: ['ocpp1.6'], // server accepts ocpp1.6 subprotocol
});

server.on('client', async (client) => {

    try {
        const response = await client.call('RemoteStartTransaction', {
            connectorId: 1, // start on connector 1
            idTag: 'XXXXXXXX', // using an idTag with identity 'XXXXXXXX'
        });

        if (response.status === 'Accepted') {
            console.log('Remote start worked!');
        } else {
            console.log('Remote start rejected.');
        }

    } catch (err) {
        console.log('Remote start errored:', err.message);
    }

});
    
server.listen(80); // listen on port 80
```

zelix25, 2022-11-23:
Thank you for this quick response. 
Indeed I am on ocpp 1.6.
I will try with your code thank you very much

zelix25, 2022-11-24:
It's good, thank you 👍

closed 2022-11-24 by zelix25