mikuso/ocpp-rpc #23: Error with remote command SendLocalList

zelix25, 2022-11-29:
Hello,
I have a problem with the SendLocalList command. I think my json structor is good but the function returns me an error.
Here is the code of my function

```
 const SendLocalList = await client.call("SendLocalList", {
       "listVersion": 1,
       "localAuthorisationList": [
        {
          "idTag":"044943121F1D80",
           "idTagInfo":{
            "expiryDate":"2023-02-01T15:09:18Z",
            "parentIdTag":"",
            "status":"Accepted",
            }
         }],
        "updateType":"Full"
});
```
Error message:
```

  rpcErrorMessage: 'Payload is syntactically correct but at least one field contains an invalid value',
  rpcErrorCode: 'PropertyConstraintViolation',
  details: {
    errors: [
      {
        instancePath: '',
        schemaPath: '#/additionalProperties',
        keyword: 'additionalProperties',
        params: { additionalProperty: 'localAuthorisationList' },
        message: 'must NOT have additional properties'
      }
    ],
    data: {
      listVersion: 1,
      localAuthorisationList: [
        {
          idTag: '044943121F1D80',
          idTagInfo: {
            expiryDate: '2023-02-01T15:09:18Z',
            parentIdTag: '',
            status: 'Accepted'
          }
        }
      ],
      updateType: 'Full'
    }
  }
}
```

mikuso, 2022-11-29:
Try using "localAuthorizationList" instead of "localAuthorisationList" (note the "z" vs the "s" in the spelling.)

zelix25, 2022-11-29:
I really didn't pay attention...
Thanks you very much :)

closed 2022-11-29 by zelix25

mikuso, 2022-11-29:
No problem.  The spec document also makes the same mistake in one place :)