mikuso/ocpp-rpc #24: How am I able to trigger an remote start and stop?

rowellcodizal, 2022-12-05:
Hi, 

I have a question,

I have this scenario 1 server(Websocket), 1 client(Charging Station) and an application(HTML implementation) that will trigger to the client.
In ocpp-rpc what is the specific syntax that I will use in my application to trigger the action like remote start and stop.?

mikuso, 2022-12-09:
Hi,

I am assuming you are using ocpp-rpc for your server implementation and that you have read through the [OCPP spec documents](https://www.openchargealliance.org/downloads/).

To give you an example of how to do this with OCPP 1.6, you would need to:
* Send a RemoteStartTransaction request to the client (charging station)
* Handle the StartTransaction message which will follow if the charging station successfully managed to start a transaction.

This is slightly different for OCPP 2.0.1.

Here's an example of this in code:

```js
const { RPCServer } = require('ocpp-rpc');

const server = new RPCServer({
    protocols: ['ocpp1.6'], // server accepts ocpp1.6
});

server.on('client', async (client) => {
    
    // handle StartTransaction requests
    client.handle('StartTransaction', ({params}) => {
        // the charging station has started a transaction and wants to inform the server.
        console.log(`Charging station started charging using ID Tag: ${params.idTag}`);

        return {
            idTagInfo: {
                status: 'Accepted', // idTag accepted
            },
            transactionId: 1, // the transactionId should relate to a record stored somewhere in your back-end
        };
    });

    try {
        // ask the client (charging station) to start a transaction, if possible
        const result = await client.call('RemoteStartTransaction', {
            connectorId: 1, // transaction should be on connector 1
            idTag: 'XXXXXX' // use an idTag with the token "XXXXXX"
        });

        console.log(`Remote start: ${result.status}`); // either "Accepted" or "Rejected"
    } catch (err) {
        console.log(`Remote start errored, because:`, err);
    }

});

server.listen(80);
```

rowellcodizal, 2022-12-11:
Hi @mikuso ,

yes I am using ocpp-rpc, but what am I trying to achieve is like I have a chat messaging app that I can send a message from client 1 to client 2.

I want to know how am I able to trigger an action from client 1 to client 2 using this ocpp-rpc because when I try to connect to the same url and identity, I am not able to achieve that scenario.

TIA

mikuso, 2022-12-11:
ocpp-rpc isn't a peer-to-peer technology, so client-to-client messaging isn't possible - you would need to route any messages through a central server.

Also, it is typically not expected for 2 different clients to connect to an endpoint with the same identity.

Perhaps if you could share some code which would better help me understand what you are trying to do, I could help further.

Thanks.

rowellcodizal, 2022-12-11:
Hi @mikuso ,

Sorry, I created confusion what am I trying to do is to trigger from HTML app to client(charging station) #2 or HTML app to server to client(charging station) #2. 

For your reference this is my current setup,
HTML App, Server, Client #1 and Client #2.
Client is my charging station

mikuso, 2022-12-11:
ocpp-rpc isn't currently usable within web browsers. I've listed some reasons for this in #19 

In order to trigger a flow like you want, you would probably want the web page to make an XHR/fetch request to a REST endpoint on your server. This endpoint would then forward that request onto your OCPP client (charging station).

rowellcodizal, 2022-12-11:
Do you have any recommendation that I can use to do that or do you have a step by step to create the REST endpoint on my server?

mikuso, 2022-12-12:
You could try following a tutorial like this: https://www.tutorialspoint.com/expressjs/expressjs_restful_apis.htm

rowellcodizal, 2022-12-12:
Do you have a sample implementation that has a server and 2 client (charging station) connected then I can trigger using the rest api to start the transaction specific to Client 1?

Sorry for this because I am new to this implementation of OCPP and Nodejs.
Thank you for your understanding.

rowellcodizal, 2022-12-12:
Hi @mikuso I think I resolve the issue using your suggestion.
Script below is only what I need for my web application, thank you so much.

app.get('/:id', (req,res) => {
    server._clients.forEach(function each(clients) {
        console.log(clients);
        if(clients.identity == req.params.id){
            try{
                clients.call('RemoteStartTransaction', {
                    idTag: "xxxx",
                    connectorId: 1
                });
            }
            catch (err) {
                res.send(err.message);
            }
        }
        else{
            res.send('Accepted');
        }

    });
});