mikuso/ocpp-rpc #25: Issue on the data return

rowellcodizal, 2022-12-22:
Hi @mikuso ,

Can you help me with my issue regarding this line of code.

When I try this, the TransactionIdentity in the return function is always zero, even though I assign the TRANSACTIONID of the query return.
Could you please assist me in resolving this? 
`var TransactionIdentity = 0
        client.handle('StartTransaction', ({params, signal, messageId}) => {
            console.log(`Server got start from ${client.identity}:`, JSON.stringify(params));
            //if(JSON.stringify(params) != ''){}
            var requestTransactionId = new sql.Request();

            requestTransactionId.query(`EXEC SP_EVO_CreateOrUpdate_RemoteStart '${client.identity}', ` + params.connectorId + `,'` + params.idTag + `', ` + params.meterStart + `,'` + params.timestamp + `'`, function(err,records){
                if(err) {console.log(err)}
                else {

                    console.log(records.recordset[0])
                    TransactionIdentity = records.recordset[0].TRANSACTIONID;
                    
                    server._clients.forEach(function each(clientrecord) {
                        if(clientrecord.identity == client.identity + 'USER' + params.connectorId){
                            clientrecord.call('StartTransactionResponse', {"timeStamp": params.timestamp})
                        }
                    });
                }
            })
            const stringdata = '{"expiryDate": "2022-11-02T12:35:24Z", "parentIdTag": "", "status": "Accepted"}';
            
            return {
                "idTagInfo": JSON.parse(stringdata),
                "transactionId": TransactionIdentity
            }

            

        });`

rowellcodizal, 2022-12-22:
Also, additional details are that when I tried to place the return function inside the query function, I got a continuous start.

mikuso, 2022-12-22:
The return statement is completing before the `requestTransactionId.query` callback is called.  Therefore, there is no opportunity for `TransactionIdentity` to change to the value from your query, as it has already returned.

You can avoid this problem by wrapping your callback into a promise and awaiting the result of it.

What SQL module are you using? It probably supports Promises natively, which would make this very easy.

rowellcodizal, 2022-12-22:
Do you have any syntax or sample code that I can use?
I use SQL Server 2018.

mikuso, 2022-12-22:
Is this the SQL module you are using? https://www.npmjs.com/package/mssql

rowellcodizal, 2022-12-22:
Sorry, this is what I use https://www.npmjs.com/package/msnodesqlv8

mikuso, 2022-12-22:
Ok, can you try doing something like this? I have re-written your code slightly.

The change I have made is to wrap your query callback into a Promise and await on the result. Note I also tagged the handler function as an async function.

```js
var TransactionIdentity = 0;
client.handle('StartTransaction', async ({params, signal, messageId}) => {
    var requestTransactionId = new sql.Request();

    var remoteStartRecord = await new Promise((resolve, reject) => {
        requestTransactionId.query(`EXEC SP_EVO_CreateOrUpdate_RemoteStart '${client.identity}', ` + params.connectorId + `,'` + params.idTag + `', ` + params.meterStart + `,'` + params.timestamp + `'`, function(err,records){
            if (err) {
                reject(err)
            } else {
                resolve(records.recordset[0]);
            }
        });
    });
    
    console.log(remoteStartRecord)
    TransactionIdentity = remoteStartRecord.TRANSACTIONID;
    
    server._clients.forEach(function each(clientrecord) {
        if(clientrecord.identity == client.identity + 'USER' + params.connectorId){
            clientrecord.call('StartTransactionResponse', {"timeStamp": params.timestamp})
        }
    });
    
    const stringdata = '{"expiryDate": "2022-11-02T12:35:24Z", "parentIdTag": "", "status": "Accepted"}';
    
    return {
        "idTagInfo": JSON.parse(stringdata),
        "transactionId": TransactionIdentity
    }
});
```

Let me know if this works for you.

rowellcodizal, 2022-12-26:
Ow, I forgot to close this, this help me with my issued encounter. thank you @mikuso

closed 2022-12-26 by rowellcodizal