mikuso/ocpp-rpc #27: Do you have an idea to get the battery percentage of the connected EV?

rowellcodizal, 2023-01-03:
Hi @mikuso ,

Do you have any idea or currently develop on how am I going to get the current battery percentage of the connected EV in the charging points?

TIA.

mikuso, 2023-01-03:
In OCPP1.6, the battery percentage is reported in the `SoC` measurand of a MeterValues request.

I believe this feature is only possible for DC charging (Mode 4).  There is no way to read a battery percentage via OCPP from an AC charger, as far as I'm aware, as the underlying charging protocol and hardware doesn't support this.

Hope this helps.

rowellcodizal, 2023-01-10:
Sorry for the late reply, but I think you are right about this battery percentage. Thank you, @mikuso

closed 2023-01-10 by rowellcodizal