mikuso/ocpp-rpc #28: Issue on the certificate.

rowellcodizal, 2023-01-10:
Hi @mikuso ,

I am asking for help with the issue that I encounter when I am trying to move to another server.

I have the .pfx file from another server but when I try to use and change to https the created server, I always have this error.
[image]

Do you have an idea how to fix this?

rowellcodizal, 2023-01-10:
Additional question, when I try to use the http but instead of localhost I use the domain to connect I have the encounter the same error.?

mikuso, 2023-01-10:
Does the .pfx include a private key? Could it be a CORS issue?

To be honest, this kind of question is outside of the scope for this module. You may have better luck asking on stackoverflow.

closed 2023-01-16 by rowellcodizal