mikuso/ocpp-rpc #29: Issue upon disconnect

rowellcodizal, 2023-01-19:
Hi @mikuso ,

do you have an idea on what we can do if the websocket server is disconnected?
Is it possible to trigger an stop to the machine or any other function of server before it is suddenly disconnected to the server.?

rowellcodizal, 2023-01-19:
Also @mikuso do you have an idea or code on how I am going to get the data from the machine?

mikuso, 2023-01-19:
> Hi @mikuso ,
> 
> do you have an idea on what we can do if the websocket server is disconnected? Is it possible to trigger an stop to the machine or any other function of server before it is suddenly disconnected to the server.?

Are you implementing a charging station or a central system (or both?)

> Also @mikuso do you have an idea or code on how I am going to get the data(Charging Transaction) from the machine?

Do you mean in the instance that the charger goes offline? I assume you are already able to get the transaction data while the charging station is online?

rowellcodizal, 2023-01-19:
> Are you implementing a charging station or a central system (or both?)

Both.

> Do you mean in the instance that the charger goes offline? I assume you are already able to get the transaction data while the charging station is online?

Regarding this, I have the scenario of when there is ongoing charging, then the connection to websocket in the machine is suddenly disconnected, and when the connection is being established, the ongoing charging was done a minute ago.
Is the charging machine capable of passing that transaction if it successfully connects to the web socket again? 

We just want in this scenario to properly recorded the exact time and meter when the charging is done. 

TIA.

mikuso, 2023-01-19:
> > Are you implementing a charging station or a central system (or both?)
> 
> Both.

Ok. In OCPP1.6 there is no built-in way to force the charger to stop charging in the event that it goes offline. If you wanted to add this kind of functionality, you could extend your implementation to support this behaviour. The simplest option might be just to add a new type of configuration key to control this. If you decide to do this, please try to ensure the name won't conflict with any other configuration key which could potentially be added to future revisions of the spec (e.g. by prefixing your product name at the start of the key name).

> Regarding this, I have the scenario of when there is ongoing charging, then the connection to websocket in the machine is suddenly disconnected, and when the connection is being established, the [...]

This is documented in the spec.  Under section 3.7 of the OCPP1.6 spec, it details exactly what the charger should do in this situation. This should solve your problem.

Excerpt from the spec:

> # 3.7. Transaction-related messages
> The Charge Point SHOULD deliver transaction-related messages to the Central System in chronological order as soon as possible. Transaction-related messages are StartTransaction.req, StopTransaction.req and periodic or clock-aligned MeterValues.req messages.
>
> **When offline, the Charge Point MUST queue any transaction-related messages that it would have sent to the Central System if the Charge Point had been online.**
>
> In the event that a Charge Point has transaction-related messages queued to be sent to the Central System, new messages that are not transaction-related MAY be delivered immediately without waiting for the queue to be emptied. It is therefore allowed to send, for example, an Authorize request or a Notifications request before the transaction-related message queue has been emptied, so that customers are not kept waiting and urgent notifications are not delayed.
>
> The delivery of new transaction-related messages SHALL wait until the queue has been emptied. This is to ensure that transaction-related messages are always delivered in chronological order.
>
> **When the Central System receives a transaction-related message that was queued on the Charge Point for some time, the Central System will not be aware that this is a historical message, other than by inference given that the various timestamps are significantly in the past. It SHOULD process such a message as any other.**
>
> ## 3.7.1. Error responses to transaction-related messages
> It is permissible for the Charge Point to skip a transaction-related message if and only if the Central System repeatedly reports a 'failure to process the message'. Such a stipulation is necessary, because otherwise the requirement to deliver every transaction-related message in chronological order would entail that the Charge Point cannot deliver any transaction-related messages to the Central System after a software bug causes the Central System not to acknowledge one of the Charge Point’s transaction-related messages.
>
> What kind of response, or failure to respond, constitutes a 'failure to process the message' is defined in the documents OCPP JSON Specification and OCPP SOAP Specification.
>
> The number of times and the interval with which the Charge Point should retry such failed transaction-related messages MAY be configured using the TransactionMessageAttempts and TransactionMessageRetryInterval configuration keys.
>
> When the Charge Point encounters a first failure to deliver a certain transaction-related message, it SHOULD send this message again as long as it keeps resulting in a failure to process the message and it has not yet encountered as many failures to process the message for this message as specified in its TransactionMessageAttempts configuration key. Before every retransmission, it SHOULD wait as many seconds as specified in its TransactionMessageRetryInterval key, multiplied by the number of preceding transmissions of this same message.
>
> As an example, consider a Charge Point that has the value "3" for the TransactionMessageAttempts configuration key and the value "60" for the TransactionMessageRetryInterval configuration key. It sends a StopTransaction message and detects a failure to process the message in the Central System. The Charge Point SHALL wait for 60 seconds, and resend the message. In the case when there is a second failure, the Charge Point SHALL wait for 120 seconds, before resending the message. If this final attempt fails, the Charge Point SHOULD discard the message and continue with the next transaction-related message, if there is any.

rowellcodizal, 2023-01-19:
Hi @mikuso,

This document means that the charging point will send the charging transaction to the central systems upon reconnecting. Is that correct?

Also, I have a question: is the stop transaction, meter value, or status notification being sent to the central system after the machine is connected to it, or is there a handler that I need to include in my code to receive the record from that transaction?

mikuso, 2023-01-19:
> Hi @mikuso,
> 
> This document means that the charging point will send the charging transaction to the central systems upon reconnecting. Is that correct?

Yes.

> 
> Also, I have a question: is the stop transaction, meter value, or status notification being sent to the central system after the machine is connected to it, or is there a handler that I need to include in my code to receive the record from that transaction?

You are mostly correct, yes.

When the machine connects back to the central system after being offline, it should send all the transaction-related messages which it was unable to send while it was offline.

These transaction-related messages are, specifically:

- **StartTransaction** requests
- **StopTransaction** requests
- clock-aligned **MeterValues** requests (i.e. not all MeterValues requests - only those which would be sent due to clock-aligned metering configuration.)

You should not queue up **StatusNotification** requests when offline. These are not considered to be "transaction-related messages".  They should simply be discarded if you can't immediately send them.

There is no special handler or logic for handling these queued transaction-related messages. As per the spec document:

> ... the Central System will not be aware that this is a historical message, other than by inference given that the various timestamps are significantly in the past. **It SHOULD process such a message as any other.**

rowellcodizal, 2023-01-19:
I have another question, @mikuso: does the changing point have the ability to stop automatically based on the dynamically set duration time?

If possible, do you have a sample code that I can use for this? 

Thanks

mikuso, 2023-01-19:
> Edited: Additional question: What would be the first trigger that the charging point is being sent to the central system? Is it the queued request or the bootnotification?

The BootNotification request should only be sent after the charge point is (re)booted.  If the charge point is simply reconnecting to the server after a temporary network outage, and has not actually rebooted, then it should not send a BootNotification when it reconnects.

In the instance that the charge point has been rebooted, then it should first send a BootNotification and wait for this to be accepted by the central system before sending anything else (including transaction-related messages).  This is written in the OCPP1.6 spec, section 4.2:

> The Charge Point SHALL send a BootNotification.req PDU each time it boots or reboots. Between the physical power-on/reboot and the successful completion of a BootNotification, where Central System returns Accepted or Pending, the Charge Point SHALL NOT send any other request to the Central System. This includes cached messages that are still present in the Charge Point from before.

mikuso, 2023-01-19:
> I have another question, @mikuso: does the changing point have the ability to stop automatically based on the dynamically set duration time?

This is not a standard feature of OCPP1.6.

Since you're implementing both the client and server, you could possibly implement this yourself using DataTransfer requests to add your own functionality.

rowellcodizal, 2023-01-24:
Thank you @mikuso this solve my issue that when the charging point is reconnect to the server the ongoing/queued transaction is sync when it is connected thanks..

closed 2023-01-24 by rowellcodizal