mikuso/ocpp-rpc #30: Is there a way to get client.identity within a handler?

VincentWillats, 2023-01-23:
I am looking for a way to access the identity within a function, currently I am wrapping the handlers like below, but I feel like there should be an easier way?
```
client.handle("StartTransaction", ({params})=>{
    return StartTransaction({params}, client.identity)
})
```

mikuso, 2023-01-23:
Hi @VincentWillats ,

Sorry, perhaps I don't understand exactly what you're trying to do - but isn't the `client` already in scope anyway?

I assume you have some code that looks like this (simplistic example):

```js
async function StartTransaction({params}, clientIdentity) {
    // register the transaction parameters and get a transaction ID to return
    const transactionId = await transactionDatabase.register(
        clientIdentity,
        params.connectorId,
        params.idTag,
        params.meterStart
    )

    // return the transaction ID
    return {
        transactionId,
        idTagInfo: { status: 'Accepted' }
    }
}

client.handle("StartTransaction", ({params})=>{
    return StartTransaction({params}, client.identity)
})
```

This could be re-written as simply:

```js
client.handle("StartTransaction", async ({params})=>{
    // register the transaction parameters and get a transaction ID to return
    const transactionId = await transactionDatabase.register(
        client.identity,
        params.connectorId,
        params.idTag,
        params.meterStart
    )

    // return the transaction ID
    return {
        transactionId,
        idTagInfo: { status: 'Accepted' }
    }
})
```

Does this help?

VincentWillats, 2023-01-23:
Apologies in my example that would be right yes, but in my actual code base I have the functions split out in other files, since adding all the logic for 1.6 and 2.0.1 in that single file is a bit bulky.

mikuso, 2023-01-23:
I see; So your `StartTransaction` function is imported from another file.

In order to avoid "forwarding" these calls one by one to each function, you might want to use a wildcard handler in an arrangement like so...

```js
const BootNotification = require('./bootnotification.js')
const StatusNotification = require('./statusnotification.js')
const StartTransaction = require('./starttransaction.js')
const StopTransaction = require('./stoptransaction.js')

const handlers = {
    BootNotification,
    StatusNotification,
    StartTransaction,
    StopTransaction,
}

client.handle(({method, params}) => {
    if (!handlers.hasOwnProperty(method)) {
        throw createRPCError("NotImplemented")
    }

    const handler = handlers[method]
    return handler({params}, client.identity)
})
```

VincentWillats, 2023-01-24:
That is a much cleaner way, thank you for the suggestion.

closed 2023-01-24 by VincentWillats