mikuso/ocpp-rpc #31: No response after call of GetConfiguration

rowellcodizal, 2023-01-24:
Hi @mikuso Any idea why I don't receive any response after I call the GetConfiguration from the central system to the charge point?

mikuso, 2023-01-24:
Can you provide some code demonstrating the issue?

rowellcodizal, 2023-01-24:
> Can you provide some code demonstrating the issue?

This is the function that I want to call but I don't get any response after calling that syntax .

client.call('GetConfiguration', {});

mikuso, 2023-01-25:
Sorry that's really not enough to go on.

Can you try this and tell me what output you get?

```js
client.call('GetConfiguration', {})
    .then(r => console.log('RESULT:', r))
    .catch(e => console.log('ERROR:', e));
```

rowellcodizal, 2023-01-26:
> Sorry that's really not enough to go on.
> 
> Can you try this and tell me what output you get?
> 
> ```js
> client.call('GetConfiguration', {})
>     .then(r => console.log('RESULT:', r))
>     .catch(e => console.log('ERROR:', e));
> ```

Hi @mikuso,

This is the output but the other configuration key is missing on this list.
[image]

I tried doing this but still the output is the list

mikuso, 2023-01-26:
Sorry I must have misunderstood. I thought the problem was that you did not get any response when making that call.  But the problem instead seems to be that the response does not contain a specific key/value pair that you were expecting to get?

Which key/value pair are you trying to get?  Am I right thinking it is "QrCode"?

This is a non-standard key (not defined in any specification I'm aware of), so won't be present in all charging stations.  If you are implementing this charging station yourself, then it will obviously be your responsibility to send this in your response to a GetConfiguration request.

I hope this helps?

liketurbo, 2023-02-02:
I tried this call:
```
client.call("GetConfiguration", {
  key: "ConfigurationVersion"
});
```

And it's caused my chargepoint to restart 🤔

mikuso, 2023-02-02:
@liketurbo Sorry but it's very hard to gauge what's going on without a bit more code and an error log.

One thing that stands out to me from the code you've provided is that your request params are not valid for that OCPP1.6 call - so if you are running in strict mode, you will get an error (and maybe a crash if you're not handling that error).

If you want to provide a "key" parameter in a GetConfiguration request, then the "key" value must be an array of strings (not a single string).

Try this instead?

```js
client.call("GetConfiguration", {
  key: [
    "ConfigurationVersion"
  ]
});
```

liketurbo, 2023-02-03:
> @liketurbo Sorry but it's very hard to gauge what's going on without a bit more code and an error log.
>
> One thing that stands out to me from the code you've provided is that your request params are [...]

Yep, that seems did the trick 🙂

closed 2023-02-21 by mikuso