mikuso/ocpp-rpc #32: Http server

zelix25, 2023-01-30:
Hello, I created a ws.js file with the server.on function and an https.js with express to send requests with the post method. I'm having trouble creating my query. I can't get the client.call function outside of my ws.js file.

My ws.js

`
const { RPCServer, createRPCError } = require('ocpp-rpc');
const {errorCodes, messageTypes, logging, port} = require('../constants/ocppConstants');
const Str = require('@supercharge/strings');
const db = require("../models");

const server = new RPCServer({
    protocols: ['ocpp1.6'], // server accepts ocpp1.6 subprotocol
    strictMode: false,       // enable strict validation of requests & responses
});

server.on('client', async (client) => {
    client.handle(({method, params}) => {
        // This handler will be called if the incoming method cannot be handled elsewhere.
        console.log(`[${new Date().toLocaleString("fr-FR")}] <<< ${method} reçus du CP #${client.identity}`);
        // throw an RPC error to inform the server that we don't understand the request.
        throw createRPCError(errorCodes.notImplemented);
    });

});

console.log(`WS start on port ${port}`);
server.listen(port);

module.exports = server;
`

my http.js
`
const express = require('express');
const server = require('../servers/ws');
const { RPCServer } = require('ocpp-rpc');

const router = express.Router();

router.post('/reset/:cp', async function(req, res, next) {

    const cp = req.params.cp;
    console.log(server._clients);

};
`

Another option would be to retrieve the session id in _session: { sessionId: '9ZpMTM5ReTQdBchPaf0w' }, to create my sending request

Thank you :)

mikuso, 2023-01-30:
Are you trying to make a call to a charging station based on its connected identity?

You may want to do something like I wrote in #26.

Try adapting your http.js as follows:

```js
const express = require('express');
const server = require('../servers/ws');

const router = express.Router();

// when clients connect (after auth), store them in a Map based on their identity
const clients = new Map();
server.on('client', client => {
    clients.set(client.identity, client);

    // when clients disconnect, remove them from our Map
    client.once('close', () => {
        // check that we're about to delete the correct client
        if (clients.get(client.identity) === client) {
            clients.delete(client.identity);
        }
    });
});

router.post('/reset/:cp', async function(req, res, next) {

    const cp = req.params.cp;
    const client = clients.get(cp);
    const result = await client.call('Reset', {type: 'Hard'});
    res.json({result});

};
```

zelix25, 2023-01-30:
Wow! already an answer!! 
Ok it's good :)
Thanks you very much :)

closed 2023-01-30 by zelix25