mikuso/ocpp-rpc #35: socket disconnects quite often

dyaacov, 2023-02-22:
Trying my luck here,
The Charger disconnects quite often, I can't even complete full cycle.

any idea what to check? how to debug?

Thanks
Dekel

mikuso, 2023-02-22:
Sure, I'd be happy to help.

Am I right in assuming that the charger is a third-party product, and you are building a Central System using this module?

What charger are you using?

What do you mean by "full cycle"?

dyaacov, 2023-02-22:
More info:
it looks like the Charger keeps sending meterValues so it means it is still connected.
but somehow, there is not client connected on the server so I can push him messages

`
rpcServer.on('client', (client) => {
    if (client.identity === 'evcharger') {
      Logger.error('No identity was specified');
      // no identity was specified
      return;
    }
    clients.set(client.identity, client);
    client.once('close', (event) => {
      Logger.info('got close event', event);
      // check that we're about to delete the correct client
      if (clients.get(client.identity) === client) {
        clients.delete(client.identity);
      }
    });
    ChargePointEventsHandler(client, handlers);
  });
`

dyaacov, 2023-02-22:
few problems:

1. I get bootNotification ans StatusChange events.
2. I use RemoteStartTransaction, the charger starts to charge but I do not get Charging status or any other status but Preparing
3. I get Meter values

after a minute or so, I get socket disconnect and from that point I can no longer communicate with the Charger. as far as I know in websockets, the client should re-open the connection.

Charger: Nidec

dyaacov, 2023-02-22:
I printed the message:

got close event {\n \"code\": 1006,\n \"reason\": \"\"\n}

mikuso, 2023-02-22:
Ok so I understand that after you perform a RemoteStartTransaction, you get a StartTransaction response. However, once the transaction has started, the connector does not change to the "Charging" status - is that right?

It may simply be that the charger is unable to detect a vehicle present. Typically the charger won't proceed to "Charging" until it has negotiated state C or D in the Mode 3/4 charging protocol of [IEC 61851](https://en.wikipedia.org/wiki/IEC_61851). If you don't have a vehicle to test with, then you can also simulate this charging state using specialist EV charger test equipment which can send the correct signal to the charger.

As for your disconnection problem, it looks like the close events are caused by a loss of connectivity between the charger and the central system. Is there a regular pattern to this? Can you time it? If it happens at a specific length of time after the connection is established, then it may indicate that some kind of inactivity timeout is occurring. Try logging the timestamps along with your connects/disconnects and see if there's any pattern that emerges.

Thanks.

dyaacov, 2023-02-22:
I also noticed that I do not get HeartbeatNotification.. it is configure to send every 2 minutes.
I'll try to change it to 5 seconds to see if it affect anything

mikuso, 2023-02-22:
Perhaps you are listening for the wrong event? In OCPP1.6, the name of the call is just "Heartbeat", not "HeartbeatNotification".

dyaacov, 2023-02-22:
yes yes, this is my class name :)

it takes EXACTLY 30 seconds to the charger/socket to disconnect

I think I got it... its AWS API Gateway... it has timeout after 30 seconds

mikuso, 2023-02-22:
Ah, well it sounds like you are making progress.

Since you are still early in development, can I make a couple of suggestions which may also help identify problems earlier?

1) Enable `strictMode` by passing `strictMode: true` in the RPCServer constructor (if you are not doing already). This will ensure that all the messages that you send and receive are spec-compliant and not causing any problems for you.

Along with this, set up an event handler to log whenever you do encounter a strictMode spec violation, like so:
```js
server.on('client', client => {

    client.on('strictValidationFailure', event => {
        console.log('Strict Validation Failure:', event);
    });

});
```

2) Set up an event handler to log whenever a non-conforming message is encountered from the charger. This may help you identify when the charger is doing something that it shouldn't.

Similar to the above, you can do this by listening for the `badMessage` event from the client, like so:
```js
server.on('client', client => {

    client.on('badMessage', event => {
        console.log('Bad Message:', event);
    });

});
```

dyaacov, 2023-02-22:
Nice!
thats a real shortcut rather then .then().catch() :)

dyaacov, 2023-02-22:
I think thats what causing it to stay on Preparing and not moving to Charging

[image]

mikuso, 2023-02-22:
I see the problem. You are not replying correctly to the StatusNotification request.

I assume your handler looks something like this?
```js
client.on('StatusNotification', ({params}) => {
    // do something

    // return nothing
    return;
});
```

You need to return an empty object in your response, as per the spec, like so:
```js
client.on('StatusNotification', ({params}) => {
    // do something

    // return empty object
    return {};
});
```

dyaacov, 2023-02-22:
I use this document and its not specifying what to return in StatusNotification
https://www.oasis-open.org/committees/download.php/58944/ocpp-1.6.pdf

Thanks!!!

mikuso, 2023-02-22:
I see. You are reading the first edition of OCPP1.6 spec document - quite an old document now.

I would recommend going to the [official OCA website](https://www.openchargealliance.org/downloads/) (the authors of the OCPP spec) to download the latest version of the OCPP1.6 spec document (edition 2) as well as the OCPP1.6-J spec (the RPC protocol which this module implements), a security whitepaper, and additional errata documents that they have published since then.  These documents contain a lot of clarifications about particular things in the OCPP spec which may not be very clear in the earlier version.

The OCA website will unfortunately ask you for your email address before sending you a download link, but I have found it to be worthwhile to get access to this wealth of information. (It is free)

---

On the specific point of the correct return value for StatusNotification, I think you are right that this is not mentioned in the OCPP1.6 core spec, and may be open to some interpretation.

The OCPP1.6-J spec does say that when a response value is not required, then either an empty object or `null` would be acceptable responses, but it does also say that returning an empty object is the preferred way to respond.

Despite this, the official OCPP1.6-J schema definitions actually turn this recommendation into an explicit hard requirement instead.  So, in line with these schema definitions (which are likely to be employed by almost every charger that you interact with), I've also made this a mandatory requirement whenever you use `strictMode`, too.

I may automatically coerce empty responses into empty objects in a future update to prevent this kind of issue coming up again, but I'll have to consider if this would have any other implications first.

dyaacov, 2023-02-22:
Websockets by default has no retry.
Does the spec forces EV Charger manufacturers to handle retries?

mikuso, 2023-02-23:
OCPP1.6 doesn't dictate exactly how a charging station should retry a websocket connection after connection loss, but it is expected that it should try to do so, yes. The OCPP1.6-J spec does describe how a charger and central system should handle BootNotifications in a reconnect scenario (as opposed to a reboot), but that's about it.

However, the OCPP-J spec is further fleshed-out in OCPP2.0.1 part 4, which includes much clearer instructions on how and when a charging station should retry its websocket connection. It says:

> When the connection is lost, the Charging Station SHALL try to reconnect. When reconnecting, the Charging Station SHALL use an increasing back-off time with some randomization until it has successfully reconnected. This prevents an overload of the CSMS when all Charging Stations reconnect after a restart of the CSMS.
> 
> The first reconnection attempts SHALL be after a back-off time of: **RetryBackOffWaitMinimum** seconds, plus a random value with a maximum of **RetryBackOffRandomRange** seconds. After every failed reconnection attempt the Charging Station SHALL double the previous back-off time, with a maximum of **RetryBackOffRepeatTimes**, adding a new random value with a maximum of **RetryBackOffRandomRange** seconds to every reconnection attempt. After **RetryBackOffRepeatTimes** reconnection attempts, the Charging Station SHALL keep reconnecting with the last back-off time, not increasing it any further.

When used in a client context, this **ocpp-rpc** module automatically uses a sensible default way to reconnect to a central system, using an exponential backoff strategy. This can be configured to your liking by specifying a `backoff` option in the RPCClient constructor. This, in turn, makes use of the [ExponentialStrategy](https://github.com/MathieuTurcotte/node-backoff#class-exponentialstrategy) defined in the [backoff](https://npmjs.com/package/backoff) module.

closed 2023-03-22 by mikuso