mikuso/ocpp-rpc #36: How to use OCPP server and charging points

juliieenn, 2023-02-22:
Hi everyone,
I saw the group is active do I try my luck.

I'm novice in github. I don't know how you install the server, and how add CP to make communication between them.
I downloaded and extracted all your files, run with CMD "npm install ocpp-rpc" but I don't know how to start the server.
Someone can help me please ? 
Thanks a lot in advance !

mikuso, 2023-02-23:
Hi, welcome to github!

Unfortunately this software is not a ready-to-go charging station management system. This is simply just a module that can be used to help when programming either a charging station or a management system using Node.js - it handles the tricky parts of the underlying communications protocol so that you can focus on implementing the commands and logic.

If this sounds what you are looking for, then you are in the right place. You'll first need to download a copy of the OCPP specification documents from the [OCA website](https://www.openchargealliance.org/downloads/) to learn how the protocol works.

Alternatively, if you are looking for a ready-built OCPP service, you might want to check out [SteVe](https://github.com/steve-community/steve). It wasn't made using this module and it's not related to this project, but it's the most feature-complete open source OCPP central system implementation that I'm aware of.

juliieenn, 2023-02-23:
Hi,
Thanks a lot for your answer ! 
Yeah actually I'm looking for an OCPP open source central system and differents CP simulator to understand how it works with OCPP1.6J. I saw SteVe server and I will install it soon.  
I don't know if I can connect other CP simulator about other authors to SteVe server (with node-js).

mikuso, 2023-02-23:
Regarding charging station simulators - I have found that most of the open source simulators are pretty poor implementations; I haven't seen any that I would feel confident in recommending to others; However, you may have better luck finding one than me.

Your best bet, if you can afford it, is to buy the [OCTT (the official OCPP Test Tool)](https://www.openchargealliance.org/protocols/test-tool/) from the OCA. I believe it is quite expensive though.

If you are just looking for an overview of how the protocol works and what it's capable of, then I recommend first looking through the [OCPP specification documents](https://www.openchargealliance.org/downloads/).

juliieenn, 2023-02-23:
Yeah Official OCPP test tool is very expensive. 
I have already a part (not all i think) of OCPP specifications documents.
I'm just looking for an OCPP server implemented with Node-js and a multiple chargings stations simulator. 
All to see how it works and to test all functions that a CP can have.

closed 2023-02-23 by juliieenn