mikuso/ocpp-rpc #37: Deviation from OCPP specification

agruenb, 2023-02-25:
First of all, thank you for the cool project. However I think I found a slight deviation from the OCPP 1.6 specification. Under _4.1.1. Synchronicity_ it says: 

> A Charge Point or Central System SHOULD NOT send a CALL message to the other party unless all the CALL messages it sent before have been responded to or have timed out.

So setting the `callConcurrency` on client or server to anything different from 1 would violate the spec.

I could see someone not as familiar with the spec increase this option to improve performance in a test environment. However in the wild a microcontroller might run into unforseen problems when receiving multiple messages.

But maybe I understood this wrong which could also be the case 😛

mikuso, 2023-02-25:
Hi, thanks for the kind words. 😊

You are correct, a call concurrency above 1 is not allowed in the current versions OCPP, but I specifically wanted to allow the option to support concurrency so that this module could be more useful for other applications.  Personally, I use this module for other IoT projects which are not related to EV charging, and this is very useful for that.

I think the best thing for me to do would be to add a note/warning in the readme advising against using a number above 1, as you are correct this can cause compatibility issues, race conditions, etc...

ZOUHIR23, 2023-04-25:
Hello dear @mikuso 

please I have a question about your solution when I do the test with a real charging station it can not connect via websocket communication I do not know is what because of localhost I use or something else if you want to help me 
thank you in advance