mikuso/ocpp-rpc #39: Websocket server does not connect with react charge point

YassineEddyb, 2023-04-26:
Hello Mikuso thanks for this amazing project.

When connecting to the server from a simulator it works fine, But when trying to connect to a real Charge Station It doesn't work, even though they are in the same network and I used my IP address to connect to the server from the Charge Point website.

I tried to lunch the server with "0.0.0.0" but no result, but when trying the same thing on a python server it works.

mikuso, 2023-04-26:
Hi Yassine,

Sounds like a tricky one to solve, but I'll do my best to help.

What make and model of charging station are you using? Would you be able to show me how this is configured? (perhaps a screenshot of the charger's OCPP configuration interface)

Does the charging station collect any logs which might hint as to the problem?

mikuso, 2023-04-26:
If you have logs, please can you share them here. I think this would be the easiest way to diagnose this.

YassineEddyb, 2023-05-12:
Hi sorry for the late reponse

I found the problem, I was trying to connect to the charge point using ethernet cable but the server was running on WSL which does not have access to the ethernet interface.

I ran the server from powershell directly and that solved the problem, thank you.

mikuso, 2023-05-12:
Thanks for the follow-up. I'm glad you were able to figure out the problem. 👍

closed 2023-05-12 by mikuso