mikuso/ocpp-rpc #42: Crashes when trying to connect to OCTT simulator

ahsanaasim, 2023-05-18:
Hi, I have tried to connect my basic central system with the OCTT simulator. When I try to connect, the system crashes. 

Here is my configuration

[image]

I receive the following error

```

[1] /Users/workstation/Programming/Clients/COS/ocpp-node/node_modules/ocpp-rpc/lib/client.js:338
[1]                 const err = Error(reason);
[1]                             ^
[1] 
[1] Error [AbortError]: Client disconnected
```

mikuso, 2023-05-18:
Can you please provide a bit more context for the error? Are you able to share your code?

ahsanaasim, 2023-05-18:
I have commented out this line. And crash went away. 

`client.call('Say', `Hello, ${client.identity}!`);`

But now I am constantly getting this warning.

```
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (Use `node --trace-warnings ...` to show where the warning was created)
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
[1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
```

mikuso, 2023-05-18:
Regarding the first issue, this makes some sense.

"Say" is not a valid OCPP1.6 request name, so it's possible that the OCTT is disconnecting the client because of this.

The call to `client.call()` will throw an Error if the the other side disconnects before sending a reply.

You can handle this error in your code by catching the Promise returned by `client.call()`

Either like:

```js
client.call('Say', 'Hello').catch(err => {
    console.error(err);
});
```

or, if using async/await, you can catch it in the traditional way:

```js
try {
    await client.call('Say', 'Hello');
} catch (err) {
    console.error(err);
}
```

As for the memory leak warning... I'm not sure I can give you more assistance on this without first seeing your code.  Can you provide a small test case that I can run to demonstrate the problem?

I'm also interested to know which version of Node.js you are using.

ahsanaasim, 2023-05-19:
I have noticed another thing. The boot notification test case from OCTT failed. This is the log from the OCTT

```
19-May-2023 17:51:51   INFO  - [TC_CS_V16_000_BOOT] [2,"1684475511210","BootNotification",{"chargePointVendor":"OCA","chargePointModel":"OCA","chargePointSerialNumber":"GLTEST","chargeBoxSerialNumber":"GLTEST"}]
19-May-2023 17:51:51   INFO  - [TC_CS_V16_000_BOOT] Waiting started for message- call blocked
19-May-2023 17:51:51   INFO  - [TC_CS_V16_000_BOOT] [Response]
19-May-2023 17:51:51   INFO  - [TC_CS_V16_000_BOOT] [3,"1684475511210",{"status":"Accepted","interval":3000,"currentTime":"2023-05-19T05:51:51.224Z"}]
19-May-2023 17:51:51   ERROR  - [TC_CS_V16_000_BOOT] 2023-05-19T05:51:51.224Z
com.google.gson.JsonSyntaxException: 2023-05-19T05:51:51.224Z
	at com.google.gson.DefaultDateTypeAdapter.deserializeToDate(DefaultDateTypeAdapter.java:107)
	at com.google.gson.DefaultDateTypeAdapter.deserialize(DefaultDateTypeAdapter.java:82)
	at com.google.gson.DefaultDateTypeAdapter.deserialize(DefaultDateTypeAdapter.java:35)
	at com.google.gson.TreeTypeAdapter.read(TreeTypeAdapter.java:58)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.read(ReflectiveTypeAdapterFactory.java:93)
	at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.read(ReflectiveTypeAdapterFactory.java:172)
	at com.google.gson.Gson.fromJson(Gson.java:803)
	at com.google.gson.Gson.fromJson(Gson.java:768)
	at com.google.gson.Gson.fromJson(Gson.java:717)
	at com.google.gson.Gson.fromJson(Gson.java:689)
	at org.oca.ocpp.core.commons.util.json.JsonParser.asJSONObject(JsonParser.java:32)
	at org.oca.ocpp.core.endpoint.client.mgr.Jetty9CPWebSocket.response(Jetty9CPWebSocket.java:149)
	at org.oca.ocpp.core.endpoint.client.mgr.Jetty9CPWebSocket.OnWebSocketMessage(Jetty9CPWebSocket.java:82)
	at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:104)
	at java.base/java.lang.reflect.Method.invoke(Method.java:578)
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:70)
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:251)
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:65)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:243)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:150)
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:322)
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:202)
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:225)
	at org.eclipse.jetty.websocket.common.Parser.parseSingleFrame(Parser.java:259)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:460)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:441)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:311)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103)
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:336)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:313)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:171)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:129)
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:388)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:806)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:938)
	at java.base/java.lang.Thread.run(Thread.java:1589)
Caused by: java.text.ParseException: Unparseable date: "2023-05-19T05:51:51.224Z"
	at java.base/java.text.DateFormat.parse(DateFormat.java:399)
	at com.google.gson.DefaultDateTypeAdapter.deserializeToDate(DefaultDateTypeAdapter.java:105)
	... 38 more
19-May-2023 17:51:51   WARN  - [TC_CS_V16_000_BOOT] Exception while Processing response
```

And I am returning this

```
const response: IBootNotificationResponse = {
        status: BootStatus.Accepted,
        interval: 3000,
        currentTime: new Date().toISOString(),
      };
      return response;
```

So I guess the format is wrong. 

I will share my entire codebase with you. I will try to pass all the test cases of both cp and cs in this weekend.

ahsanaasim, 2023-05-19:
I have pushed my entire codebase here
https://github.com/ahsanaasim/ocpp-node

mikuso, 2023-05-19:
The error that the OCTT is giving you seems to be due to its inability to parse the date string `"2023-05-19T05:51:51.224Z"`.

This seems to me to be perfectly valid. Could this be a bug in the OCTT?

* The OCPP1.6 spec (section 3.15) states that all time notations must be in ISO 8601 format.
* `new Date().toISOString()` does create an ISO 8601 format date string ([as documented in MDN](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Date/toISOString))
* The JSON schema draft 04 (which the OCPP1.6 spec uses) also defines `date-time` fields such as this to be ISO 8601 compatible.

One thing I would be tempted to try is to see if it can handle the datetime without the fractional seconds (even though section 3.15 specifically allows this).  Try and see if this works?  `new Date().toISOString().replace(/\.\d+/, '')`

I would test this myself but I don't have access to the OCTT unfortunately.

I will have a look at your other issues when I get time later.

ahsanaasim, 2023-05-20:
I have tried your workaround. But still crashes. I have mailed oca about this. Let's see what they say. I will update you once I get a response.

ahsanaasim, 2023-05-20:
I have a question about meter values. Whenever I send a trigger message for meter values, I just get the values for the following mrasurand 

```

[{'sampled_value': [{'value': '11903.75889583333333333333', 'unit': 'Wh', 'measurand': 'Energy.Active.Import.Register'}], 'timestamp': '2023-05-10T13:38:22.000000Z'}]

```

I have noticed this for both octt and Eaton chargers. I was wondering how would I get the meter values of other types of mrasurands.

mikuso, 2023-05-20:
> I was wondering how would I get the meter values of other types of mrasurands.

Check the OCPP1.6 spec for information about the `MeterValuesSampledData` configuration key. You need to set this with the comma-separated list of measurands you want to receive.

You should also be aware that there are separate configuration keys for specifying measurands to be used in clock-aligned meter values and in StopTransaction requests.

A full list of measurands is available in section 7.31.

mikuso, 2023-05-20:
> But now I am constantly getting this warning.
> 
> ```
> [1] (node:70334) MaxListenersExceededWarning: Possible EventTarget memory leak detected. 11 abort listeners added to [AbortSignal]. Use events.setMaxListeners() to increase limit
> ```

Regarding this, I have cloned your repo, built and run your code, and connected a barebones simulator to the running server. I was not able to reproduce this issue.

Can you provide reproduction steps. Can you also advise which version of Node.js you are using?

Thanks.

ahsanaasim, 2023-05-20:
Can you share the simulator you used?

ahsanaasim, 2023-05-29:
I have a few questions regarding the meter values. Not about the library. Rather it's a question about the OCPP itself. I can create a separate thread for this if you think I should.

**Question 1**
How can I calculate the energy consumption (kwh) for a specific charging session and charging rate (kw). Which meter values measurand should I follow for this?

I am currently working with a charger from Eaton. 

**TransactionId**
I can see from the OCPP doc that, meter values have an optional field transaction id that I am setting when sending a remote start transaction request. But I have never received this transaction id in metervalues from Eaton Chargers when I trigger it from the central system. I can see there is no way to send transaction id when triggering from central system. I can pass connector id only. 

**My Assumptions**
For energy taken by a car in a particular session, I need to use "Power.Active.Import". But that is not provided by the charger.  Not sure why. I have added it in the configuration. I also noticed that the meter start value in the start transaction and meter stop value in stop transaction response both are sent 0. So I can not calculate from these 2 values. 

And, I am not sure which measurand to use for the charging rate. I am assuming Power.Active.Import. I haven't tried it. I will update it after I Try. 

As you are an expert in OCPP, so I believe you have answers to my questions and confusions. Thanks

mikuso, 2023-05-30:
It would be helpful if you could post new questions in separate issues - ideally one question per issue to make them easier for me to answer (although you may include multiple questions in one issue if they are all related).  This may also help other people find answers more easily, as well as make it easier for me to keep track of the questions. In any case, I'll do my best to answer your questions here:

> **Question 1**
> How can I calculate the energy consumption (kwh) for a specific charging session and charging rate (kw). Which meter values measurand should I follow for this?

To measure the energy consumption for the entire charging session, it is best practice to take the `meterStop` value provided by the charger in the StopTransaction message, and deduct the `meterStart` value given by the charger in its corresponding StartTransaction message. This will give you a total energy consumption for that transaction measured in watt-hours (Wh).

Divide this number by 1000 to get the total kWh value.

> I also noticed that the meter start value in the start transaction and meter stop value in stop transaction response both are sent 0. So I can not calculate from these 2 values.

This indicates that either:
* No energy (or very little energy) was drawn from the charger during the transaction.
* There is a fault with the energy reading. In these cases, it's also common to receive `StatusNotification` messages containing an `errorCode` value of `PowerMeterFailure`.

Also keep in mind that some consumer-grade chargers aren't equipped with meters precise enough to be able to report very small increments of energy usage, and so won't report any change at all until they have recorded maybe 10 Wh or 100 Wh of energy.

> I am not sure which measurand to use for the charging rate. I am assuming Power.Active.Import.

This is correct. To read the power usage (rate of charge) at any particular time during a charging session, you should read the `Power.Active.Import` measurement. Not all measurands will necessarily be supported by all chargers, but in my experience this one is widely supported.

If you are not seeing a `Power.Active.Import` reading in the MeterValues requests from the charging station, then you may need to first send a ChangeConfiguration request to change the `MeterValuesSampledData` and/or `MeterValuesAlignedData` configuration values to include the `Power.Active.Import` measurand. You can include multiple measurands by concatenating them with commas - like so:

`Energy.Active.Import.Register,Power.Active.Import`

If your ChangeConfiguration request to change the `MeterValuesSampledData` fails (i.e. response status is "Rejected"), then it indicates that one or more of the measurands are not supported by the charger (it'll require some trial-and-error to discover which one(s)).

> **TransactionId**
> I can see from the OCPP doc that, meter values have an optional field transaction id that I am setting when sending a remote start transaction request. But I have never received [...]

The `transactionId` value is returned by the server in response to the StartTransaction request from the charging station. It's not possible to set a transactionId value any other way.

When a MeterValues message is sent from the charging station to the OCPP server, it should include a transactionId if it relates to a transaction. It should only omit the transactionId field in cases where the MeterValues request is a clock-aligned reading which does not relate to any charging session, or if it's some other kind of reading unrelated to a specific connector (e.g. a temperature reading). The OCPP1.6 errata document goes into more detail explaining how the charging station should determine which transactionId a clock-aligned meter value ought to belong to.

If you find that the charging station is not including a transactionId for the MeterValues messages which ought to relate to a transaction, then I can only conclude that it is violating the spec. The transactionId is only considered "optional" in the cases where there is no relevant transactionId to report - so this is not something that can simply be omitted from an implementation.

Unfortunately, in my experience you will find that a lot of chargers will vary in how well they adhere to the OCPP spec. The only way to get some reasonable level of assurance that your charger is fully compliant is to check if it has a certification from the OCA - otherwise it's a bit of a wild west out there. You may need to contact the manufacturer of the charging station for further assistance.

closed 2023-07-17 by mikuso