mikuso/ocpp-rpc #43: Asking about the EmergencyStop reason for stop

rowellcodizal, 2023-05-29:
Hi @mikuso ,

Do you have any idea why I got this EmergencyStop after the machine started charging?
Is this an issue on the machine or in my line of code?

mikuso, 2023-05-30:
"EmergencyStop" is usually the reason given when a transaction ends due to someone pushing an emergency stop button on the charger itself. If this is happening without the button being pressed (or perhaps the charger doesn't even have such a button equipped) then this would appear to be a fault with the charger.

There isn't any mechanism for an OCPP central system to cause a charger to stop a transaction with the "EmergencyStop" reason - so this would not be caused by an error in the server code.

rowellcodizal, 2023-06-02:
Hi @mikuso ,

asking again, because when this emergencystop is reflected on the output parameter in StopTransaction the machine still continues charging, and when I manually trigger a remotestop the machine will then stop. 

Also, when they try to start the charging again, it will continue with no emergencystop reason.

mikuso, 2023-06-02:
In normal circumstances, the charger should not continue to charge after it has sent the server a StopTransaction request.

(The only time I believe this is allowed to occur is when the charger is recovering from a suspended period of time being offline and it needs to transmit the history of charging sessions that have stopped & started since it was last online. In this scenario, it's possible for the charging station to send a historical StopTransaction request associated with a prior transaction, without interrupting its current transaction.)

However, since it sounds like this behaviour is repeating (you've tested this multiple times?), then I doubt this is the case for you.  It sounds most likely that the charger you are using is simply not fully OCPP compliant.

closed 2023-07-17 by mikuso