mikuso/ocpp-rpc #44: "server" in wssOptions cannot be used.

mf-fujimori, 2023-06-06:
I want to create a server with an endpoint specified using Express
However, noServer=true, and the following error occurs even if Server is specified.

TypeError: One and only one of the "port", "server", or "noServer" options must be specified

[image]

Any solution?

mf-fujimori, 2023-06-06:
thank you. I know how to use it because I read the documentation.

I want Express to coexist with Socket.io.
Is it possible to use it like this?
`
const Express = require('express');
const Http = require('http');
const { RPCServer } = require('ocpp-rpc');
const SocketIo = require('socket.io');
const server = Http.Server(app);

const io = SocketIo(server, { path: '/yyyy' });

const rpcServer = new RPCServer({
    protocols: ['ocpp2.0.1'],
    strictMode: true,
    wssOptions: {
        server: server,
        path: '/xxxxx'
    }
});
server.listen(59742, function () {
  Log.i('Server start.');
});
`

Since the noServer property is set to true internally, it seems that it cannot be specified.

mikuso, 2023-06-07:
You can do this. You simply need to add a condition when handling the http server's `'upgrade'` event to ensure that the OCPP server only processes upgrades at path `/xxxxx`, instead of all paths (which would conflict with socket.io)

I have re-worked this sample for you:

```js
const express = require('express');
const http = require('http');
const { RPCServer } = require('ocpp-rpc');
const { Server: SocketIOServer } = require('socket.io');

const app = express();
const httpServer = http.createServer(app);
const ioServer = new SocketIOServer(httpServer, { path: '/yyyy' });

const rpcServer = new RPCServer({
    protocols: ['ocpp2.0.1'],
    strictMode: true,
});

httpServer.on('upgrade', (request, socket, head) => {
    if (request.url.startsWith('/xxxxx')) {
        // upgrade OCPP
        rpcServer.handleUpgrade(request, socket, head);
    }
});

ioServer.on("connection", (socket) => {
    console.log(`Socket.IO client connected!`, socket.id);
});

rpcServer.on('client', client => {
    // OCPP client connected
    console.log('OCPP client connected!', client.identity);
});

httpServer.listen(59742);
```

I hope this helps.

mf-fujimori, 2023-06-08:
Thank you so much for your polite answer.

closed 2023-06-08 by mf-fujimori