mikuso/ocpp-rpc #46: Check if client is connected

sfantinati, 2023-06-08:
hello Mikuso, how can i check if a client   connection is still active or closed? 
I tested with .close but it doesn't work.

Thank you

mikuso, 2023-06-08:
You could check if `client.state === 3` [[See docs here](https://github.com/mikuso/ocpp-rpc#clientstate)].

Perhaps though, it would be better if you listen for the client's [`'close'` event](https://github.com/mikuso/ocpp-rpc#event-close-1) instead. This seems to be more useful in more situations.

Also, in case this helps you... If you are looking for an easy way to get a reference to an online client, [you may want to try doing something like this](https://github.com/mikuso/ocpp-rpc/issues/26#issuecomment-1380034985).

sfantinati, 2023-06-08:
oh very nice :) 
Thank you very much

closed 2023-06-08 by sfantinati