mikuso/ocpp-rpc #49: Asking about How to request something from ocpp-server

HaseaGhebyon, 2023-07-12:
Sorry for asking things if that are already in the document. But I'm still confused. The client will call the server using client.call, and the server will handle it via the client event. 
But how do you request something from the server and how do you handle it on the client?
I asking this with purpose to implements operations that initiated by Central System, such as : 
- Cancel Reservation
- Change Availability
- Change Configuration
- etc

mikuso, 2023-07-17:
Sorry, I'm not sure exactly what you're asking.

Are you creating a charging station or a central system?

If you're just looking for a way to reference a connected client connected to an RPCServer instance (in order to make a request), then take a look at this example: https://github.com/mikuso/ocpp-rpc/issues/26#issuecomment-1380034985

mikuso, 2023-08-31:
Closing stale issue.

closed 2023-08-31 by mikuso