mikuso/ocpp-rpc #51: Unable to set chargingProfileKind.

Seekay369, 2023-07-17:
const chargingProfile = {
        chargingProfileId: 1,
        stackLevel: 1,
        chargingSchedule: [
            {
                startPeriod: 0,
            },
            {
                startPeriod: 10,
            },
        ],
        chargingProfileKind: ocpp.ChargingProfileKind.Relative,
        ChargingProfilePurposeType: ocpp.ChargingProfilePurposeType.TxProfile
    };
    
    Kindly help how to set chargingProfileKind to Relative? Can it be string "Relative"?

mikuso, 2023-07-17:
I'm not sure if you're in the right place. `ocpp.ChargingProfileKind` is not an export defined by this library.

Can you provide more context?

Seekay369, 2023-07-17:
Sorry that I provide not enuf information.
For my situation, I want to set chargingProfile in RemoteStartTransaction.req.
According to OCPP1.6 protocol, We are able to set chargingPoint.

[image]

[image]
Inside the chargingProfile consist of chargingProfilePurpose, chargingProfileKind, chargingSchedule which is required field.

My question is how to set chargingProfilePurpose, chargingProfileKind, chargingSchedule by using this library.
Below are my code.

async function startChargingWithProfile(clientId) {
    const chargingProfile = {
        chargingProfileId: 1,
        stackLevel: 1,
        chargingProfileKind: 'Relative',
        chargingProfilePurposeType: 'TxProfile'
    };
    console.log('Received start charging');
    console.log('Client ID: ' + clientId);
    const sampleClient = clients.get(clientId);
    console.log('Sample Client: ' + sampleClient);
    if (sampleClient) {
        const response = await sampleClient.call('RemoteStartTransaction', {
            connectorId: 1, // start on connector 1
            idTag: 'seekaypersonal', // using an idTag with identity 'XXXXXXXX'
            chargingProfile: chargingProfile,
        });

        if (response.status === 'Accepted') {
            console.log('Remote start worked!');
        } else {
            console.log('Remote start rejected.');
        }

        return response.status;
    }
    else {
        return 'Client is not available.'
    }
}

mikuso, 2023-07-17:
> My question is how to set chargingProfilePurpose, chargingProfileKind, chargingSchedule by using this library.

There are no enum-like constants - you simply need to write the string values.

Seekay369, 2023-07-17:
is this okay if I set chargingProfileKind to 'Relative' and 'TxProfile' in string value?
Because it is not allow me to specify the start time and duration of the charging schedule relative to the start of the transaction, eventhrough the transaction is sucess.

`async function startChargingWithProfile(clientId) {
    
    const chargingSchedulePeriod = {
        startPeriod: 0,
        limit: 8.1,
    }
    
    const chargingSchedule = {
        duration: 60,
        chargingRateUnit: 'W',
        chargingSchedulePeriod: [
            chargingSchedulePeriod
        ]
    }

    const chargingProfile = {
        chargingProfileId: 1,
        stackLevel: 1,
        chargingProfileKind: 'Relative',
        chargingProfilePurposeType: 'TxProfile',
        chargingSchedule: chargingSchedule,
    };
    
    const sampleClient = clients.get(clientId);
    if (sampleClient) {
        const response = await sampleClient.call('RemoteStartTransaction', {
            connectorId: 1, // start on connector 1
            idTag: 'seekaypersonal', // using an idTag with identity 'XXXXXXXX'
            chargingProfile: chargingProfile,
        });

        if (response.status === 'Accepted') {
            console.log('Remote start with profile worked!');
        } else {
            console.log('Remote start with profile rejected.');
        }

        return response.status;
    }
    else {
        return 'Client is not available.'
    }
}`

Seekay369, 2023-07-21:
Thanks for explaining all these. Appreciated.

closed 2023-07-21 by Seekay369