mikuso/ocpp-rpc #57: Wildcard handler

Baladon23, 2023-08-04:
https://github.com/mikuso/ocpp-rpc/blame/c499f8a96960f993fa7b285f437185b9e03bca50/lib/client.js#L258

Shouldn't this be something like `if (handler instanceof Function && !method) { this._wildcardHandler = handler; }`

mikuso, 2023-08-04:
No, the `handle()` method has a different signature depending on whether you are setting a wildcard handler or not.

The docs should be clear that the first argument (method) is optional, in which case if you pass only a handler to `handle()`, then it will be received as the "method" parameter instead.

I hope that makes sense.  It's a pretty common pattern in js libraries where the function signature has overrides where the first parameter(s) are optional.

closed 2023-08-04 by mikuso

Baladon23, 2023-08-04:
The documentation was super clear (I probably overlooked that it is `[method,]` and not `method,`), thank you very much, everything makes perfect sense now. 
Though I still find it somewhat unintuitive that in `this._wildcardHandler = method`, based on the variable names and the function header alone, a string is assigned to a handler. But if this is standard then I accept it. 
Thank you again for the explanation.