mikuso/ocpp-rpc #58: Module Not Found using Next JS

HaseaGhebyon, 2023-08-09:
I've got this problem when integrate my client to Next JS
`Module not found: Can't resolve 'timers/promises'

https://nextjs.org/docs/messages/module-not-found`

HaseaGhebyon, 2023-08-09:
After seeing this...
https://github.com/nodejs/node/issues/39407

I realized in documentation that 
This module is built for Node.js and does not currently work in browsers.

But as we know, Next JS provide server rendering. Can this be a solution to use this/your OCPP module?

HaseaGhebyon, 2023-08-09:
Maybe its too late if i asking this, but can i build EV Charger simulator (web based) using your module?

mikuso, 2023-08-09:
> But as we know, Next JS provide server rendering. Can this be a solution to use this/your OCPP module?

I am not familiar with Next JS so I cannot assist you with this. The module was not designed to be compatible with Next JS.

> Maybe its too late if i asking this, but can i build EV Charger simulator (web based) using your module?

No. You'd need to run the simulator using Node.js and then serve up an interface to control the simulator using an (non-OCPP) API - perhaps also using Websockets if you want real-time communications.

closed 2023-08-09 by mikuso

HaseaGhebyon, 2023-08-10:
Thanks for your help. Its answering my confusion. Really appreciate for this module😃