mikuso/ocpp-rpc #60: How do StartTransactions work

ArthurHoeke, 2023-08-12:
I was wondering, if I connect my charger to the OCPP server and plug the charger into my car will it always automatically sent out a StartTransaction call or does it by default always first need to be initiated via a remote start? I want to create an authentication system, which would require a user to first plug in the charging cable to their car and request a remote start transaction call via the OCPP server to initiate charging, passing a unique user idtag for authentication which gets verified in 'StartTransaction'.

How would this be best achieved? If the charger automatically does a startTransaction request upon connecting to the car, would best practice be to always verify the idTag? Or should user authentication be done differently?

Thank you!

mikuso, 2023-08-12:
The charger does not typically send a StartTransaction upon connecting to the EV.

Instead, once the cable is connected, the charger would normally send a StatusNotification request with the "status" of "Pending" to indicate that the charging station is now "occupied" (- although this is a bit of a loose term).

For the charger to actually send a StartTransaction message to the central system, you'd typically need to first either send a RemoteStartTransaction from the server to the charger, or perform some physical action locally at the charger (- like wave an RFID card in front of it).  Prior to the StartTransaction request, the charger may also send an Authorize request to confirm that the provided idTag is valid before starting the transaction (this behaviour is configurable).

There are a few configuration options to tweak this flow slightly - but this is the gist of it.  (You may find that different chargers have all manner of non-standard configuration options that can change this flow to work in other ways, - such as the one you described in your question).

I hope this helps somewhat.

ArthurHoeke, 2023-08-12:
> The charger does not typically send a StartTransaction upon connecting to the EV.
>
> Instead, once the cable is connected, the charger would normally send a StatusNotification request with the [...]

Absolutely, thanks for taking the time to reply!

closed 2023-08-12 by ArthurHoeke