mikuso/ocpp-rpc #61: Correct way to send a handshake or connect

Tarekmo110, 2023-08-13:
I'm trying to simulate a charge point and connect to the server. I think the server is working fine but the problem is how i connect to it. 
I'm using this to connect 
const newSocket = new WebSocket(`${ocppJEndpointUrl}/${encodeURIComponent(chargePointIdentity)}`,['ocpp1.6', 'ocpp1.5']);
so it should look like this 
ws://localhost:3000/ABCD with protocols [ocpp1.6,ocpp1.5]
this is the error i get 

> const err = Error(reason);
                            ^
Error [AbortError]: Client disconnected
    at Object.pendingCall.abort (/home/tarek/Desktop/ocpp_csms/server/node_modules/ocpp-rpc/lib/client.js:338:29)
    at /home/tarek/Desktop/ocpp_csms/server/node_modules/ocpp-rpc/lib/client.js:440:63
    at Array.forEach (<anonymous>)
    at RPCServerClient._rejectPendingCalls (/home/tarek/Desktop/ocpp_csms/server/node_modules/ocpp-rpc/lib/client.js:440:48)
    at RPCServerClient._handleDisconnect (/home/tarek/Desktop/ocpp_csms/server/node_modules/ocpp-rpc/lib/client.js:458:14)
    at WebSocket.<anonymous> (/home/tarek/Desktop/ocpp_csms/server/node_modules/ocpp-rpc/lib/client.js:410:49)
    at Object.onceWrapper (node:events:629:26)
    at WebSocket.emit (node:events:526:35)
    at WebSocket.emit (node:domain:489:12)
    at WebSocket.emitClose (/home/tarek/Desktop/ocpp_csms/server/node_modules/ws/lib/websocket.js:258:10)

mikuso, 2023-08-13:
Hi,

Please can you clarify a few things for me?

- Are you using this module to create a central system, a charging station simulator, or both?
- Is the error being reported by your client (simulator) or by the server?

Thanks.

Tarekmo110, 2023-08-13:
I will use it to create a central system but i was just testing things out first and familiarizing myself with this module.

I have a simple ui with a connect button that tries to establish a web-socket connection with the server i created using the example you gave in express 

I get this in the browser [https://imgur.com/a/bzsPmKk](url)

And I get this in the server [https://imgur.com/a/WWyW64S](url) I added some console.log to client.js 

if u need any more info please let me know

mikuso, 2023-08-13:
From your screenshots, it looks like the server is crashing, which is why your client's websocket is disconnecting.

Can we take a look at your server code to see what might be causing that to happen?