mikuso/ocpp-rpc #65: Central system microservice

inane, 2023-10-03:
Hello

I will try to explain my scenario. I have one frontend app with different buttons that are sending actions to the api-gateway and from there to communicate with the central system developed with your library and the chargepoints, listening a worker queue based in BullMQ.

I think I am not being able understand correctly the configuration and the differences client / server and call/handle to use it on premise.

Here a sample of my chaotic that is working but not with the on-premise requirements.

`import {RPCClient, RPCServer} from 'ocpp-rpc';
import express from "express";

const app = express();
const router = express.Router();
app.use(router);

const cli = new RPCClient({
  endpoint: 'ws://whatever:9002/eosp/ws/charge_points', // the OCPP endpoint URL
  identity: 'test-mar',             // the OCPP identity
  protocols: ['ocpp1.6'],          // client understands ocpp1.6 subprotocol
  strictMode: true,                // enable strict validation of requests & responses
});

// connect to the OCPP server
await cli.connect();
const allClients = new Map();

// send a BootNotification request and await the response
const bootResponse = await cli.call('BootNotification', {
  chargePointVendor: "Chargevite",
  chargePointModel: "Model-S",
});
console.log('boot response :', bootResponse)
if (bootResponse.status === 'Accepted') {
  console.log('Server accepted the client!');
  // send a Heartbeat request and await the response
  const heartbeatResponse = await cli.call('Heartbeat', {});
  // read the current server time from the response
  console.log('Server time is:', heartbeatResponse.currentTime);
}
const authNotification = await cli.call('Authorize', {
  idTag: 'R0O2YFNPG2RJX4G0WZJD'
});

console.log('info tag authorized :', authNotification)

const startTransaction = await cli.call('StartTransaction', {
  connectorId: 1, // start on connector 1
  idTag: '123456789', // using an idTag with identity 'XXXXXXXX'
  meterStart: 0,
  timestamp: new Date().toISOString()
});
console.log('startTransaction :', startTransaction)
const rpcServer = new RPCServer(
  {
    protocols: ['ocpp1.6'], // server accepts ocpp1.6 subprotocol
    strictMode: true
  }
);

rpcServer.on('client', async (client) => {
  console.log('client :', client)
  allClients.set(client.identity, client); // store client reference
  console.log('allClients :', allClients)
  client.handle('StartTransaction', rpcServer.handleUpgrade)
  client.once('close', () => {
    // check that we're about to delete the correct client
    if (allClients.get(client.identity) === client) {
      allClients.delete(client.identity);
    }
  });
});

export default rpcServer
`
I hope your clarifications to handle correctly my purpose.

Thank you in advance

mikuso, 2023-10-03:
Hi there.

Can you please clarify whether you are trying to implement a Central System, or a Charging Station? Your code seems to indicate that you are trying to implement both at the same time.

inane, 2023-10-03:
WSUrl is connecting to one central system based in Python. ChargePoints connect to that CS via websocket. My plan now is to use the library to connect with the current CS (python) to test responses from the charge points on premise, because current CS has not endpoints to handle requests  on premise after click one button for instance.

In a future, I would like to change that backend in python and to handle it via microservice.

Here one example

[](https://imgur.com/a/Ub5kRnW)

Sorry for my bad explanation

mikuso, 2023-10-03:
Sorry I'm struggling to understand. I think we have a language barrier problem.

My understanding is that you are trying to create a Charging Station (or at least, a simulator of sorts) to connect to a pre-existing Central System written in Python.

In this case, you ought to start by removing the RPCServer references from your code. You do not need this.

Your next step is probably to add some method handlers to the RPCClient instance?

e.g.
```js
cli.handle('RemoteStartTransaction', ...
```

mikuso, 2023-10-03:
The image shows three different components to a system. Which of these are you trying to implement here? Which do you need help with?

inane, 2023-10-03:
Charge points will be connected to the central system. Currently, devices are connected to the central system based in python. I would like to change central system part based in python by new implementation based in JS.

Charger points are connected via websockets to the current system. But These chargers should be connected to the new ws url (if I am not wrong) in this new development.

these are the different parts 🙏🏻

mikuso, 2023-10-03:
Ok, I see. So you are creating a Central System?

I notice in your example that you are attempting to handle a StartTransaction request by using the rpcServer's `handleUpgrade` callback. This is not correct. You should write your own callbacks to handle each OCPP method. Follow the example in the readme.

closed 2023-10-19 by inane