mikuso/ocpp-rpc #67: Example for using this package in a project that uses ESM module

xiaojieTao, 2023-12-05:
There is no out of box example for using this package in a project that uses ESM module so I edited one of the example in the README:

```
import express from "express";
let rpc;
async function load() {
    rpc = await import("ocpp-rpc");
};

(async () => {
    await load();
    const app = express();
    const httpServer = app.listen(3000, 'localhost');

    const rpcServer = new rpc.RPCServer();
    httpServer.on('upgrade', rpcServer.handleUpgrade);

    rpcServer.on('client', client => {
        // RPC client connected
        client.call('Say', `Hello, ${client.identity}!`);
    });

    // create a simple client to connect to the server
    const cli = new rpc.RPCClient({
        endpoint: 'ws://localhost:3000',
        identity: 'XYZ123'
    });

    cli.handle('Say', ({params}) => {
        console.log('Server said:', params);
    });

    await cli.connect();
})();
```

mikuso, 2023-12-05:
Why not just write `import { RPCServer } from 'ocpp-rpc';` ?

I'm not sure what the async import is for.

xiaojieTao, 2023-12-05:
> Why not just write `import { RPCServer } from 'ocpp-rpc';` ?
> 
> I'm not sure what the async import is for.

I am beginner to node and JS. Some post from stack overflow said I need to use dynamic import to import CommonJS module into ESM project. I guest either that info is outdated or ocpp-rpc is compatible with both CommonJS and ESM. Is ocpp-rpc a CommonJS module?