mikuso/ocpp-rpc #69: Correct way to set the maximum watt a charger can charge in a RemoteStartTransaction.

villszer, 2024-01-28:
Hello and thank you for this library. What i don't understand is how to set a maximum amper or watt a charger should charge in a remote request. Currently i have a test environment where a node.js server ( with express ) communicates with a (Mennekes) charger.

Here is my test method for starting a charge ( it is triggered by a client side button click )

```js

// Generate  a dummy profile of 1 watt limit and max 5 sec charging.

const generateChargingProfile = function(){
    const dummyProfile= {
        chargingProfileId: 1,
        stackLevel: 1,
        validFrom: new Date().toISOString(),
        validTo: new Date(Date.now() + 1 * 60000).toISOString(),
        chargingProfilePurpose: 'TxDefaultProfile',
        chargingProfileKind: "Relative",
        chargingSchedule: {
            chargingRateUnit: "W",
            chargingSchedulePeriod: [
              {
                limit: 1,
                startPeriod: 0
              }
            ],
            duration: 5
        }
    };
    console.log("Charging profile: ", JSON.stringify(dummyProfile, null, 2));
    return dummyProfile;
}

const startCharging = async function({identity,connectorID}){
    let charger = await Charger.findOne({identity: client.identity}); // Get the charger from db
    if( !charger ){ return false; }
    if( !charger.isActive ){ console.log("Charger is not active!"); return false; } // Check if the charger is active.
    const client = clients.get(identity); // Get the client instance
    log(client,"RemoteStartTransaction",{identity,connectorID}); // log the request
    const response = await client.call('RemoteStartTransaction', { // Generate a request and wait for a response.
        connectorId: parseInt(connectorID),
        idTag: dummyIDTag,
        chargingProfile: generateChargingProfile(),
    });
    log(client,"RemoteStartTransactionResponse",response);
    return response.status === 'Accepted';
}
```

So in here as you can see i generate a dummy profile and i expect the charger to charge a maximum 1 watt power. But it just does not stop. I have a car simulator equipment which simulates a car when i plug it in the charger. I also have a regular 220v lamp plugged in so it can draw power from the charger. Everything is working, the charger sends an Authorize request, my server accepts it and the charger starts the charing. When i wait for the validTo property ( it is 1 min from the current time ) or for the duration in the schedule or for the 1 watt limit i set it doesnt stop. If i send a RemoteStopTransaction command from the server to the charger it just rejects it.

What is wrong? I want to tell the charger to charge only a specific amount and no more.

mikuso, 2024-01-29:
Hi @villszer,

This question is a little out of scope for this module, but I'll try my best to help anyway.

I don't think it's possible to set an energy cap using OCPP1.6 - It's not part of the protocol spec as far as I'm aware. If the charger supports setting an energy cap, then it would require using a proprietary method specific to that charger (and therefore won't be applicable in the general sense to other chargers).

To help clear up any confusion, I'll also just mention that the charging profile you've created actually works more like a speed limit than a cap. Limiting of Amps or Watts will only slow down how fast energy is transferred to the vehicle, not limit the total amount of energy that it is allowed to be transferred. You are probably mixing up Watts with Watt-Hours (Wh). Watts is a measurement of power, whereas Watt-Hours is a measurement of energy. You cannot set a cap of Watt-Hours using a charging profile.

The best way I've found to enforce an energy cap is to monitor the MeterValues requests coming from the charger until it shows that the amount of energy received by the vehicle has exceeded the limit you want to set. At this point, you should issue a RemoteStopTransaction to stop the charger.

If you are having issues with RemoteStopTransaction, feel free to post some example code and I'll try to point out where you may be going wrong.

Thanks,

villszer, 2024-01-30:
Thank you very much for the clarification. Meanwhile i read the exact same thing on reddit as you just told me now.
I think it is a really poor design by the OCPP guys. If the server can't set a maximum amount of power draw for the charger and the server is responsible for the stopping of the transaction, what happens when the server goes down after a RemoteStartTransaction? What happens if there are massive packet losses or network error. What happens if the sim card in the charger goes out of internet? The charger just charges indefinietly? What is most ridiculus is that the charger what i test ( Mennekes ) is locks the cable if a transaction starts and wont let it until it stops. But if the server experiences some error and can't stop the charging then the cp wont let the cable. That is ridiculus.

I tought that the charger will stop the transaction after a specified amount of power drawn.

But if this is the case, i have some serious problems because my charger rejects the RemoteStopTransaction after the session is started and doesn't even send MeterValues.

My use case is just the following

Server sends RemoteStartTransaction -> Charger sends Authorize -> Server accepts Authorize -> Charger sends StartTransaction -> Server accepts StartTransaction -> Charging starts for real -> Charger charges indefinietly -> RemoteStopTransaction rejected -> No MeterValues has been sent.

Thank you for your kind help.

villszer, 2024-01-30:
I have strict mode enabled and here are my functions

```js
const startCharging = async function({identity,connectorID}){
    let charger = await Charger.findOne({identity: identity});
    if( !charger ){ return false; }
    if( !charger.isActive ){ console.log("Charger is not active!"); return false; }
    const client = clients.get(identity);
    log(client,"RemoteStartTransaction",{identity,connectorID});
    const response = await client.call('RemoteStartTransaction', {
        connectorId: parseInt(connectorID),
        idTag: dummyIDTag,
        chargingProfile: generateChargingProfile(),
    });
    log(client,"RemoteStartTransactionResponse",response);
    return response.status === 'Accepted';
}

const stopCharging = async function({identity,connectorID}){
    let charger = await Charger.findOne({identity: identity});
    if( !charger ){ return false; }
    if( !charger.isActive ){ console.log("Charger is not active!"); return false; }
    const client = clients.get(identity);
    log(client,"RemoteStopTransaction",{identity,transactionId:dummyTransactionID});
    const response = await client.call('RemoteStopTransaction', {
        transactionId: dummyTransactionID,
        //idTag: dummyIDTag,
        //connectorId: parseInt(connectorID)
    });
    log(client,"RemoteStopTransactionResponse",response);
    return response.status === 'Accepted';
}

const startTransaction = function(client,params){
    log(client,"StartTransaction",params);
    const expiryDate = new Date(Date.now() + 1 * 60000).toISOString();
    return {
        status: "Accepted",
        transactionId: dummyTransactionID,
        idTagInfo: {
            status: "Accepted",
            expiryDate: expiryDate,
        },
    };
}

const stopTransaction = function(client,params){
    log(client,"StopTransaction",params);
    clearInterval(meterValueRequestTimer);
    return {
        status: "Accepted",
        idTagInfo: {
            status: "Accepted",
        },
    }
}

const bootNotification = async function(client,params){
    log(client,"BootNotification",params);

    let charger = await Charger.findOne({identity: client.identity});
    if( charger ){
        charger.isActive = true;
        charger.set({ ...charger._doc, ...params });
        const resp = await charger.save();
        if( resp ){
            events.sendAll("BootNotification",charger);
            return {
                status: "Accepted",
                interval: 20, // sec
                currentTime: new Date().toISOString()
            };
        }
    }
    return {
        status: "Rejected",
        currentTime: new Date().toISOString()
    };
}

const authorize = function(client,params){
    log(client,"Authorize",params);
    return {
        idTagInfo:{
            status: "Accepted"
        }
    };
}
```

This is a test environment, the server accepts everything now.

villszer, 2024-01-30:
Here is some log info from the charger. It looks like the buffer is fully filled and can't send requests to the server

```
Jan 30 19:50:12 cp cp_app[955]: <41>                 websock_comm.c:909  [01cca000] (*D) - websocket_msg_received len:424 = '[2,"47b7ee9e-6ccc-413c-8b67-c1bf2614f297","RemoteStartTransaction",{"connectorId":1,"idTag":"testIDTag","chargingProfile":{"chargingProfileId":1,"stackLevel":1,"validFrom":"2024-01-30T19:50:13.460Z","validTo":"2024-01-30T19:52:13.460Z","chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"limit":16,"startPeriod":0}],"duration":1000}}}]' 
Jan 30 19:50:12 cp cp_app[955]: <41>                 websock_comm.c:723  [01cca000] (*D) - websocket_msg_received(): COMMAND, processing... 
Jan 30 19:50:12 cp cp_app[955]: <41>                    json_comm.c:624  [01cca000] (*D) - json_comm_parse_ts(): failed to parse datetime '2024-01-30T19:50:13.460Z', retrying with '2024-01-30T19:50:13Z' 
Jan 30 19:50:12 cp cp_app[955]: <41>                    json_comm.c:627  [01cca000] (*D) - json_comm_parse_ts(): successfully parsed fixed datetime '2024-01-30T19:50:13Z' 
Jan 30 19:50:12 cp cp_app[955]: <41>                    json_comm.c:624  [01cca000] (*D) - json_comm_parse_ts(): failed to parse datetime '2024-01-30T19:52:13.460Z', retrying with '2024-01-30T19:52:13Z' 
Jan 30 19:50:12 cp cp_app[955]: <41>                    json_comm.c:627  [01cca000] (*D) - json_comm_parse_ts(): successfully parsed fixed datetime '2024-01-30T19:52:13Z' 
Jan 30 19:50:12 cp cp_app[955]: <41>                 websock_comm.c:734  [01cca000] (*D) - websocket_msg_received(): successfully parsed COMMAND (3), notifying... 
Jan 30 19:50:12 cp cp_app[955]: <41>                  ocpp_msging.c:138  [01cca000] (  ) - OCPP RemoteStartTransaction.req received [2,"47b7ee9e-6ccc-413c-8b67-c1bf2614f297","RemoteStartTransaction",{"connectorId":1,"idTag":"testIDTag","chargingProfile":{"chargingProfileId":1,"stackLevel":1,"validFrom":"2024-01-30T19:50:13.460Z","validTo":"2024-01-30T19:52:13.460Z","chargingProfilePurpose":"TxProfile","chargingProfileKind":"Absolute","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"limit":16,"startPeriod":0}],"duration":1000}}}] 
Jan 30 19:50:12 cp cp_app[955]: <41>                  ocpp_server.c:69   [01cca000] (*I) - ocpp_server_handle_backend_command(): command 0x1e55190 confirmation 0x1e436e0 
Jan 30 19:50:12 cp cp_app[955]: <41>                  ocpp_server.c:182  [01cca000] (*I) - ocpp_server_handle_backend_command(): Received OCPPRemoteStartTransaction 
Jan 30 19:50:12 cp cp_app[955]: <20>              command_handler.c:584  [01cca000] (*I) - ocpp_connector_handle_remote_start_transaction_command(): Connector id: 1, idtag: testIDTag, from_master = 0, no_ocpp_start_stop = 0 
Jan 30 19:50:12 cp cp_app[955]: <20>              command_handler.c:614  [01cca000] (*I) - ocpp_connector_handle_remote_start_transaction_command(): Rejecting command (neither in Idle nor Reserved state) 
Jan 30 19:50:12 cp cp_app[955]: <41>                 websock_comm.c:2221 [01cca000] (*D) - websock_comm_send_message(): processing CONFIRMATION... 
Jan 30 19:50:12 cp cp_app[955]: <41>                  ocpp_msging.c:142  [01cca000] (  ) - OCPP RemoteStartTransaction.conf sent [3, "47b7ee9e-6ccc-413c-8b67-c1bf2614f297", {"status": "Rejected"}] 
Jan 30 19:50:12 cp cp_app[955]: <41>                 websock_comm.c:497  [01cca000] (*D) - calling lws_write() with 67 bytes and content [3, "47b7ee9e-6ccc-413c-8b67-c1bf2614f297", {"status": "Rejected"}] 
Jan 30 19:50:17 cp cp_app[955]: <06>                charging_info.c:153  [01cca000] (  ) - Charge duration: 208 sec(s) 
Jan 30 19:50:17 cp cp_app[955]: <06>                charging_info.c:110  [01cca000] (  ) - Charging Info (EV): I: 0.00/0.00/0.00 A, P: 0 W, E:40712 Wh, U: 244/0/0 V 
Jan 30 19:50:20 cp cp_app[955]: <09>                        meter.c:2660 [01cca000] (*I) - meter_parse_measurands(): added measurand 'Energy.Active.Import.Register', 0 
Jan 30 19:50:20 cp cp_app[955]: <09>                        meter.c:2663 [01cca000] (*I) - meter_parse_measurands(): num 1 
Jan 30 19:50:20 cp cp_app[955]: <41>          meter_value_request.c:735  [01cca000] (*D) - create_meter_value_request(): getting 3-phase Energy from meter failed 
Jan 30 19:50:20 cp cp_app[955]: <41>          meter_value_request.c:459  [01cca000] (*I) - meter_value_create_unsigned_value_1x(): reading context:'Sample.Periodic' time '1706644220' 
Jan 30 19:50:20 cp cp_app[955]: <41>                  ocpp_client.c:1513 [01cca000] (*I) - send_request_to_backend(): (OCPP MeterValues mode:queued persistent not transaction relevant) 
Jan 30 19:50:20 cp cp_app[955]: <41>              persistent_list.c:622  [01cca000] (*W) - persistent_list_add_elem(): list reached the maximum size of 100 
Jan 30 19:50:20 cp cp_app[955]: <41>                  ocpp_client.c:1531 [01cca000] (*W) - send_request_to_backend(): message_id 59590155 cannot be queued - dropping! 
Jan 30 19:50:20 cp cp_app[955]: <21>          modem_state_monitor.c:200  [01cca000] (*I) - URC: +CREG?^M (Registration status) 
Jan 30 19:50:25 cp cp_app[955]: <06>                   statistics.c:48   [01cca000] (*I) - MK_PIC24 statistics - RX: 1648 (6.837 kB), TX: 1013 (3.774 kB), CRC16 errors: 0, non-recoverable comm errors: 0 
[... cut ...]
Jan 30 19:53:30 cp cp_app[955]: <41>                  ocpp_client.c:1513 [01cca000] (*I) - send_request_to_backend(): (OCPP MeterValues mode:queued persistent not transaction relevant) 
Jan 30 19:53:30 cp cp_app[955]: <41>              persistent_list.c:622  [01cca000] (*W) - persistent_list_add_elem(): list reached the maximum size of 100 
Jan 30 19:53:30 cp cp_app[955]: <41>                  ocpp_client.c:1531 [01cca000] (*W) - send_request_to_backend(): message_id 109417096 cannot be queued - dropping! 
Jan 30 19:53:37 cp cp_app[955]: <06>                charging_info.c:153  [01cca000] (  ) - Charge duration: 408 sec(s) 
```

mikuso, 2024-01-31:
> Thank you very much for the clarification. Meanwhile i read the exact same thing on reddit as you just told me now. I think it is a really poor design by the OCPP guys. If the server can't set a [...]

I agree. This does appear to be a big oversight in the spec. I believe there are a couple of ways you could implement charger-side time limits (although support for this is spotty), but not charger-side energy limits. 😕

> But if this is the case, i have some serious problems because my charger rejects the RemoteStopTransaction after the session is started and doesn't even send MeterValues.

Based on the limited information available, I can only guess as to why the RemoteStopTransaction is being rejected. Perhaps the issue is that you are re-using transaction IDs that have previously been used. This can confuse some chargers. See the OCPP1.6 errata document v4, section 3.54:

_"It is RECOMMENDED to use unique positive numbers for transactionIds. Negative numbers and zero (0) are possible, but
don’t have any special meaning to the Charge Point (they don’t mean the transaction is rejected or something like that.) Note
that the Charge Point might use transactionId = -1 in transaction related message when the Charge Point was not able to
successfully deliver the StartTransaction.req."_

The reasoning behind this is that chargers have their own internal memory of previous transactions, and it may be confused thinking you want to stop a transaction which has already ended - in which case, this should be rejected. It's unfortunate that the StartTransaction.conf message doesn't give the charger any opportunity to object to (or even send an error in response to) a bad transaction ID, so there's no way to be informed of a conflict.

> Server sends RemoteStartTransaction -> Charger sends Authorize -> Server accepts Authorize -> Charger sends StartTransaction -> Server accepts StartTransaction -> Charging starts for real -> Charger charges indefinietly -> RemoteStopTransaction rejected -> No MeterValues has been sent.

Looking at your logs, it does appear that there is an issue with the charger's internal buffer filling up with transactional messages. My best guess is that you don't currently have a MeterValues handler set up (or perhaps it is set up incorrectly). Chargers will buffer meter values if they cannot be sent, as they are deemed to contain important transaction-related information.

Whenever you boot-up a charger, it should attempt to send any buffered transaction-related messages to the central system before doing anything else; However, it seems that this is failing for some reason.

I recommend checking that your MeterValues handler is set up correctly, and also switch to using unique transaction IDs. This may solve your immediate problems.

Also, if you are still unable to stop a transaction, you can try issuing an UnlockConnector command to the relevant connector ID. If the charger supports this operation, then it will immediately follow up with a StopTransaction message, which will also reveal the transaction ID of the transaction that was stuck.

Hope this helps.

villszer, 2024-01-31:
Thank you for all the info!

What you say about the `transactionId` makes sense. I did reuse a `transactionId` because i had a constant defined for testing purposes. This might be the problem. I will try both of your suggestions now.

The MeterValues are setup like any other for me

```js
client.handle("MeterValues", async function({params}){
    return await handler.meterValues(client,params);
});
```

```js
const meterValues = async function(client,params){
    let charger = await Charger.findOne({identity: client.identity});
    if( !charger ){
        console.log(`[RPC] - (${client.identity}) Can't find charger in the database!`);
        return {};
    }
    //charger.connectors[params.connectorId] = {...params};
    log(client,"MeterValues",charger);
    //charger.save();
    return {};
}
```

villszer, 2024-01-31:
Here is my `StartTransaction` and my `HeartBeat`

```js
const startTransaction = function(client,params){
    log(client,"StartTransaction",params);
    const expiryDate = new Date(Date.now() + 1 * 60000).toISOString();
    currentTransactionID = generateTransactionID();
    return {
        status: "Accepted",
        transactionId: currentTransactionID,
        idTagInfo: {
            status: "Accepted",
            expiryDate: expiryDate,
        },
    };
}

const heartBeat = async function(client,params){
    log(client,"Heartbeat",params);
    let charger = await Charger.findOne({identity: client.identity});
    if( !charger ){
        return {
            status: "Rejected",
            currentTime: new Date().toISOString()
        }
    }
    charger.lastHeartBeat = new Date();
    events.sendAll("Heartbeat",charger);
    charger.save();
    return {
        status: "Accepted",
        currentTime: new Date().toISOString()
    }
}
```

villszer, 2024-01-31:
I have implemented almost every request message handlers

```js
const init = function (rpcServer) {
    rpcServer.auth(async (accept, reject, handshake) => {
        handler.auth(accept, reject, handshake);
    });

    rpcServer.on('error', (err) => {
        console.log(`[RPC] - Server error: ${err}`);
    });

    rpcServer.on('upgradeAborted', (event) => {
        console.log(`[RPC] - Server upgradeAborted:`, event);
    });

    rpcServer.once("close",function(connection){
        console.log("[RPC] - Server closed! Connection: ", connection);
    });

    rpcServer.on('client', async (client) => {
        if( !await handler.connect(client) ){ return; }
        
        client.once("close",function(){
            handler.disconnect(client);
        });

        client.once("badMessage",function(event){
            handler.badMessage(event);
        });

        client.handle("StartTransaction", function({params}){
            return handler.startTransaction(client,params);
        });

        client.handle("StopTransaction", function({params}){
            return handler.stopTransaction(client,params);
        });

        client.handle("StatusNotification", async function({params}){
            return await handler.statusNotification(client,params);
        });

        client.handle("BootNotification", async function({params}){
            return await handler.bootNotification(client,params);
        });

        client.handle("Heartbeat", async function({params}){
            return await handler.heartBeat(client,params);
        });

        client.handle("MeterValues", async function({params}){
            return await handler.meterValues(client,params);
        });

        client.handle("Authorize", function({params}){
            return handler.authorize(client,params);
        });

        client.handle( async ({ method, params }) => {
            handler.unImplemented(method, params);
        });
    });
}
```

mikuso, 2024-01-31:
Can you try setting up event listeners for the `badMessage` and `strictValidationFailure` events? You may be getting some silent failures.

I suspect something wrong with your heartbeat response because the charger's logs indicate that an InternalError (type 4 message) was received.

villszer, 2024-01-31:
I have badMessage listener

```js
 client.once("badMessage",function(event){
    handler.badMessage(event);
});

const badMessage = function(event){
    console.log("[RPC] - Bad message: ", event);
}
```

Will

villszer, 2024-01-31:
> > Strange. I updated the transactionId and now the charger started to send MeterValue requests but not StartTransaction requests.
> 
> It's possible that the charger is sending you a bunch of historical metervalues messages which previously weren't getting through.
> 
> According to the spec, the charger should send transactional-related messages in chronological order, so it's not completely unexpected in this instance to be receiving a slew of old MeterValues messages before the StartTransaction messages begin to flow again.

Yeah, the charger sends historical events but also at the specified interval which is 10sec

villszer, 2024-01-31:
Should be this way

```js
client.once("badMessage",function(event){
    handler.badMessage(event);
});

client.once("strictValidationFailure",function({params}){
    handler.strictValidationFailure(params);
});
```

or this way

```js
client.handle("badMessage",function(event){
    handler.badMessage(event);
});

client.handle("strictValidationFailure",function({params}){
    handler.strictValidationFailure(params);
});
```

mikuso, 2024-01-31:
Third way... 😄

```js
client.on("strictValidationFailure",function(event){
    console.log(event)
});
```

There's nothing that you strictly need to handle here. It's just a way of being notified of errors relating to validation.

villszer, 2024-01-31:
So should i just send back an empty object for heartBeat message?

villszer, 2024-01-31:
Should be the badMessage .on too?

villszer, 2024-01-31:
I can't find these in the docs

villszer, 2024-01-31:
Sorry for my questions. I feel really stupid to ask all of these. Thank you very much again for your help!

villszer, 2024-01-31:
For some reason i don't get StartTransaction message so i don't know the correct transactionId for RemoteStopTransaction.
I belive it should be coming from StartTransaction since i don't specify any transactionId on RemoteStartTransaction neither on Authorize.

villszer, 2024-01-31:
For HeartBeat i do this now. Omitted status from the message.

```js
const heartBeat = async function(client,params){
    log(client,"Heartbeat",params);
    let charger = await Charger.findOne({identity: client.identity});
    if( !charger ){
        console.log(`[RPC] - (${client.identity} - Heartbeat) Can't find charger in the database!`);
        return { currentTime: new Date().toISOString() }
    }
    charger.lastHeartBeat = new Date();
    events.sendAll("Heartbeat",charger);
    charger.save();
    return { currentTime: new Date().toISOString() }
}
```

mikuso, 2024-01-31:
> So should i just send back an empty object for heartBeat message?

Check the OCPP spec document. You need to send back the current time. The docs will give you the format.

> Should be the badMessage .on too?

Yes. Use "on" rather than "once", otherwise you'll only be notified of the first bad message and will ignore any others.

> I can't find these in the docs

It's all documented under the API section of the readme:

[Event: 'strictValidationFailure'](https://github.com/mikuso/ocpp-rpc?tab=readme-ov-file#event-strictvalidationfailure)
[Event: 'badMessage'](https://github.com/mikuso/ocpp-rpc?tab=readme-ov-file#event-badmessage)

> For some reason i don't get StartTransaction message so i don't know the correct transactionId for RemoteStopTransaction. I belive it should be coming from StartTransaction since i don't specify any transactionId on RemoteStartTransaction neither on Authorize.

The central system specifies the transaction ID for all transactions (even the ones started by the charger itself).

When you respond to a StartTransaction message, you provide the charger with the transaction ID in this response. The charger should remember this ID as a reference to this transaction on this connector. When you want to stop the transaction you can pass this ID with your RemoteStopTransaction request.

villszer, 2024-01-31:
Interestingly I don't get any StartTransaction request but the charing is starting anyway....
This is how i handle it now

```js
const startTransaction = function(client,params){
    log(client,"StartTransaction",params);
    const expiryDate = new Date(Date.now() + 1 * 60000).toISOString();
    currentTransactionID = generateTransactionID();
    return {
        status: "Accepted",
        transactionId: currentTransactionID,
        idTagInfo: {
            status: "Accepted",
            expiryDate: expiryDate,
        },
    };
}
```

villszer, 2024-01-31:
So UnlockConnector is worked.
Should i use this instead of RemoteStopTransaction?

mikuso, 2024-02-02:
UnlockConnector should be a last resort. It isn't the intended way for you to stop a charge.

I will note though that it is unusual that the charger didn't send you a StopTransaction and cease charging after you called UnlockConnector. This sounds like a spec violation by the charging station.

Have you managed to make any further progress?

villszer, 2024-02-02:
Yes, i did. Fortunatelly it all just clicked together.
I did as you suggested, added `badMessage` and `strictValidationFailure` handlers, adjusted some response object structure based on the `strictValidationFailure` logs and it works now.
Thank you very much for all of the help.
I really appreciate it!

closed 2024-02-02 by villszer