mikuso/ocpp-rpc #72: Disconnects after 30 sec

tmb-01, 2024-03-13:
It works fine on my local machine but when I deploy it to my VPS, WS disconnects after 30 sec. 

[image]

I thought I had a limitation in my VPS for WS but my other WS is working fine.

I am using this simulator.
https://lamerat.github.io/OCPP-Charge-Point-Simulator/

tmb-01, 2024-03-14:
@mikuso Can you help me to solve this problem?

mikuso, 2024-03-14:
Try listening on the RPCServerClient instance for the `'socketError'` and `'close'` events and log what happens.

Do you get a socket error event?
Do you get a close event? If so, what close code & reason are you given?

tmb-01, 2024-03-25:
when I start it with pm2, I face this error, but when I start it with node it works fine. But I cant use node to run in background.

closed 2024-04-14 by tmb-01