mikuso/ocpp-rpc #73: Error While Disconnecting from Webscoket Client

mardommah, 2024-03-18:
When trying to disconnect from websocket using postman or insomnia, always returning error like this
[image]

mikuso, 2024-03-18:
It looks like you're not handling a certain condition.

Can you provide a small sample of code which I can use to replicate the issue?

mardommah, 2024-03-20:
Hello, here my code example, currently im integrating it with express js  to create command initiated by central system -> charging station

**rpc.js**
[image]

**api.js**
[image]

**server.js**
[image]

mikuso, 2024-03-20:
Hi @mardommah ,

Can you please copy + paste the code so that I can try running it? I'd rather not have to re-type it.

mardommah, 2024-03-20:
okay here's the code 

**server.js**

`const express = require('express');
const serverRouter = require('./api')
const rpcServer = require("./rpc")

const app = express();
app.use(serverRouter)
const httpServer = app.listen(3002, 'localhost', () => {
    console.log("server running on 3002")
});

httpServer.on('upgrade', rpcServer.handleUpgrade);

`

**rpc.js**

`const { RPCServer } = require('ocpp-rpc');

const rpcServer = new RPCServer({
    protocols: ['ocpp1.6'],
    strictMode: false
});

rpcServer.on('client', client => {
    // RPC client connected
    client.call('Say', `Hello, ${client.identity}!`);

    // create a specific handler for handling Heartbeat requests
    client.handle('Heartbeat', ({ params }) => {
        console.log(`Server got Heartbeat from ${client.identity}:`, params);

        // respond with the server's current time.
        return {
            currentTime: new Date().toISOString()
        };

    });

    client.once('close', () => {
        // check that we're about to delete the correct client
        return "client disconnected"
    });
});

module.exports = rpcServer `

**api.js**

`const express = require('express');
const rpcServer = require("./rpc")

const serverRouter = express.Router();

serverRouter.use(express.json())

// when clients connect (after auth), store them in a Map based on their identity
const clients = new Map();

rpcServer.on('client', client => {
    clients.set(client.identity, client);

    // when clients disconnect, remove them from our Map
    client.once('close', () => {
        // check that we're about to delete the correct client
        if (clients.get(client.identity) === client) {
            clients.delete(client.identity);
        }
    });
});

serverRouter.post('/reset/:cp', async (req, res, next) => {

    const cp = req.params.cp;
    const client = clients.get(cp);
    console.log(cp)
    const result = await client.call('Reset', { type: 'Hard' });
    res.json({
        Status: "Success"
    });

});

module.exports = serverRouter`

mikuso, 2024-03-21:
Hi @mardommah ,

I wasn't able to reproduce this issue with the code you've provided. I have attempted to connect a simple client to localhost:3002 but it does not trigger the error.

Can you give me more details of how to reproduce the error you've reported, using this code?

mikuso, 2025-02-16:
Closing stale issue. If you need further help, please open a new issue. Thanks.

closed 2025-02-16 by mikuso