mikuso/ocpp-rpc #78: Unable to start server with await

neeact, 2024-06-07:
Hello,

I'm using OCPP-RPC with Node.js.
When using 
`await server.listen(3000);`

the Node.js throws the error "_await is only valid in async functions and the top level bodies of module_".

I can only start the server without await - i.e.: 
`server.listen(3000);`

Can you please help?
Thank you in advance!
-ct-

mikuso, 2024-06-07:
As per the error message you received, you'll need to use await within an async function, like so:

```js
(async function main() {
    await server.listen(3000);
})();
```

closed 2024-06-07 by mikuso

neeact, 2024-06-08:
Super!!! 
Thank you!!!
(I know this is more a Node.js question but your help is much appreciated!)

neeact, 2024-06-08:
> Super!!! Thank you!!! (I know this is more a Node.js question but your help is much appreciated!)
> 
> Actually my problem is that on the message Handlers in the OCPP server we need to communicate with our backend server and we should only "respond" to the OCPP Client after we receive the response from the backend server.
> 
> Current I'm not able to achieve this because once we receive the message from the client, the thread is closed without a response. Therefore I would need to use await to "suspend" the process until I have an answer and respond to the client afterward - i.e.:
> 
> ```
> server.on('client', async (client) => {
>   client.handle('BootNotification', ({params}) => {
>     
>       const ret = await callBackendAsync('BootNotification', client.identity, params);    
>       return ret;
>       
>   }
> }
> ```
> 
> This is not allowed by Node.js (it is returning the error "_await is only valid in async functions and the top level bodies of modules_")
> 
> I would like to avoid calling the backend server synchronously and I believe async should be used in this scenario.
> 
> Can you share some tips on how to achieve this? Thank you so much for your help once more! -ct-

I've managed to use the "await" by adding "async" to the handler function:

```
server.on('client', async (client) => {
  client.handle('BootNotification', async ({params}) => {
    
      const ret = await callBackendAsync('BootNotification', client.identity, params);    
      return ret;
      
  }
}
```

Do you see any issue with this approach?
Thank you once more for your reply.
-ct-

mikuso, 2024-06-08:
This is fine. This is how I anticipated people to use handlers like this.

neeact, 2024-06-08:
Thank you for confirming Mikuso!