mikuso/ocpp-rpc #79: Getting Client Identity from the URL and not from the handshake.identity

neeact, 2024-08-07:
Hello!

I'm having the following problem, not sure if you can help:

some OCPP devices when connecting to the server are showing an empty handshake.identity.

I understand that some devices set their OCPP identity directly in the OCPP Server Endpoint URL - i.e.:

if OCPP server URL is "ws://myocppserver.dummycom", they point the Client Device to "ws://myocppserver.dummycom/A12345", being "A12345" the intended device identity.

How can I get the Client Device Identity from the server address in alternative (or in addition) to the handshake.identity?

Any help is much appreciated!
Thank you.

gareth3wm, 2024-08-07:
Hi @neeact 

By definition (as per the OCPP docs), the device identity is basically just the tail of the path of the websocket URL.

Is there a different scenario you've encountered in the wild?

mikuso, 2024-08-07:
Hi @neeact 

By definition (as per the OCPP docs), the device identity is basically just the tail of the path of the websocket URL.

Is there a different scenario you've encountered in the wild?

neeact, 2024-08-07:
Okay, I have just found out (was not aware) that in the handshake information the "endpoint" is also filled and it contains this information - example of a handshake payload:

```
(...)
3|ksocpp  |   protocols: Set(1) { 'ocpp1.6' },
3|ksocpp  |   endpoint: '/**A12345**',
3|ksocpp  |   identity: '1523390420',
3|ksocpp  |   query: URLSearchParams {},
(...) 
```

So I can get the Client identity from the handshake.endpoint.

Thank you!

neeact, 2024-08-07:
Closing this (non) issue :)

closed 2024-08-07 by neeact

mikuso, 2024-08-07:
Ah, ok. Perfect. 👍