mikuso/ocpp-rpc #81: Help for sending commands (like pause and resume)

SKB-CGN, 2024-11-22:
Hi,
perhaps you can help me a little, in understanding, whats wrong here.

I am trying to build up a smart-charging system by my own.

I use your project to handle a Wallbox.

Currently i am struggling with the following scenario.

The car is connected to the Wallbox, the server receives the "Preparing" Status from the WB - so far so good. Now, i start the charging with:
```js
await sendCommand('RemoteStartTransaction', { connectorId: DEFAULT_CONNECTOR_ID, idTag: DEFAULT_ID_TAG });
Response: "Accepted"

```
After that, i put the car into pause via Setting a ChargeProfile with the Limit of 0 (which means 0 Amp).

The WB stops charging immediately and the status is "SuspendedEVSE".
After checking, if enough solar-energy is available, i want to start the charging again and send a new profile to the WB with a proper limit of 6A (e.g. 10A). But neither the WB changes to "charging" nor the amp is raised.

Might you be able, to have an idea here?

mikuso, 2024-11-22:
Hi,
It sounds like you are doing the correct thing.

Can you capture the OCPP logs to/from your charger - it might help identify where things are going wrong.

On the other hand, this might not be an OCPP issue at all. It could simply be that the charging station and/or the EV have an issue that prevents a charging session from resuming from 0 amps.

The logs should make things clearer, if you can provide them.

SKB-CGN, 2024-11-22:
Hi,
first of all - thanks for your reply :) This is much appreciated!

Perhaps you know the Software "EVCC", which also offers this type of charging with solar. I am trying to build that a bit smaller and they do the same thing (as far as i understand from their code).

Could you provide, how i could capture the logs?

Currently the things look like this:

**Function for "Preparing" sourced from StatusNotification:**
```js
case 'Preparing': {
   console.log('Wallbox is preparing a new car!');

   // First put the car into pause mode and check requirements
   const response = await sendCommand('SetChargingProfile', generateChargeProfile(0));
   console.log("ChargeProfile", response);

   if (response && response.status == 'Accepted') {
      // Start the Remote
      const response = await sendCommand('RemoteStartTransaction', { connectorId: DEFAULT_CONNECTOR_ID, idTag: DEFAULT_ID_TAG });
      console.log("RemoteTrans", response);
   }

   // Check, if charging is possible
}
```

**Function generateChargeProfile:**
```js
/**
 * @param {number} amps
 */
function generateChargeProfile(amps) {
   const now = new Date();
   const timestamp = new Date(now.getTime() - 60 * 1000).toISOString(); // Reduce one minute
   console.log(timestamp);

   return {
      "connectorId": DEFAULT_CONNECTOR_ID,
      "csChargingProfiles": {
         "chargingProfileId": 1,
         "stackLevel": 0,
         "chargingProfilePurpose": "TxDefaultProfile",
         "chargingProfileKind": "Absolute",
         "chargingSchedule": {
            "startSchedule": timestamp,
            "chargingRateUnit": "A",
            "chargingSchedulePeriod": [
               {
                  "startPeriod": 0,
                  "limit": amps
               }]
         }
      }
   }
}
```

mikuso, 2024-11-22:
You can log OCPP messages using this module by listening for the 'message' event, like so:
```js
client.on('message', ({message, outbound}) => {
    console.log(outbound ? '>>>' : '<<<', message);
});
```

mikuso, 2024-11-22:
Hi, perhaps there is some kind of time synchronisation issue?

Have you tried omitting the `startSchedule` property? This property is optional, and the charging station should apply the charging schedule immediately if it is omitted.

It would be good to see all of your message logs up until this point (everything for this session up to this point, BootNotification onwards). Including timestamps with each message would also help.

SKB-CGN, 2024-11-22:
Sure, will set that up again and come back to you shortly.

Thanks for your great help so far!

mikuso, 2024-11-26:
I would have expected the status of the connector to change to SuspendedEVSE once you set the 0-amp charging profile, but it remained as Charging. Perhaps this is a mistake in the Wallbox reporting that it is charging when it is not?

Try enabling periodic meter values sampling so we can see for sure whether or not the EV is charging.

To do this:
* Set the `MeterValueSampleInterval` config value to something like `30` so we can get a meter reading every 30 seconds.
* Set the `MeterValuesSampledData` config value to `"Energy.Active.Import.Register,Current.Offered,Current.Import"` so we can get readings of the current and energy flowing.

Let's see what logs we get back then.

SKB-CGN, 2024-11-26:
Thanks for coming back on this topic 👍

I think, I have had a coding error while putting the Wallbox to pause with 0 Amp.

Will verify that tomorrow again and come back.

In the meantime - what do you think, is the overall best way, to apply the first charge profile, to set the wb into pause?

I want to archive, that the wallbox will be set to pause first, every time a car connects.

After that, I will check the conditions and start charging or leave the wallbox in pause.

SKB-CGN, 2024-11-30:
Hi,
thanks for sharing all informations and thoughts about my topic. This is much appreciated ;)

I went further with all the things you provided and i found out in the meanwhile. During this period, i got confused a lot cause of the reactions of my wallbox. I think, they implemented OCPP in a "special" way, but dont support the necessary specifications.

Your library is used, to build up an adapter for the automation system iobroker - so dont wonder, why the logging or commands look a bit different ;)

Please have a look at the following things i did so far.

I am starting the server and waiting for the client to connect - this works. After that, i set some configurations, which also works.
Where i start to get confused is:

```js
await this.sendCommand('SetChargingProfile', this.generateChargeProfile(0));
```
The command JSON is:
```
>>> 30.11.2024, 11:23:18[2,"ea8bf886-4bbc-4b7f-be3a-f320bca00e95","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
```
Answer:
```
<<< 30.11.2024, 11:23:19[3,"ea8bf886-4bbc-4b7f-be3a-f320bca00e95",{"status": "Accepted"}]
```
This indicates - from my understanding, that the profile is installed successfully and set to 0A. When i have a look into the Webinterface of my Wallbox, it still says:
[image]

After that, i request the MeterValues, which returns, if no car is connected:
```
OCPP-Charge: {"chargePower":0,"chargeCurrent":32,"meterStart":0,"meterEnd":0,"session":0,"lastValue":4865260.2}
```

So, the wallbox still is in 32A mode, which is obviously not "correct".
Therefore, i need to send the below command, after the wallbox says it is in "Preparing". ```chargeAmps``` here is some variable - mostly 0 for putting the car into pause first.
```js
// First start the RemoteTransAction
const response = await this.sendCommand('RemoteStartTransaction', { connectorId: DEFAULT_CONNECTOR_ID, idTag: DEFAULT_ID_TAG });
if (response.status === 'Accepted') {
	// Set the Wallbox into Pause-Mode or charge directly; depends on status
	const response = await this.sendCommand('SetChargingProfile', this.generateChargeProfile(chargeAmps));
	this.log.info(`[OCPP]: Wallbox has accepted the remote start and is set to ${chargeAmps}A!`);
	this.log.info("[OCPP]: ChargeProfile" + JSON.stringify(response));
}
```

This puts the car into pause - but using a ```TxProfile``` here, is not working - nor sending it directly with the ```StarteRemoteTransaction```. The wallbox returns with ```Accepted``` but does not use the profile according to my testing.

So, from my point of view, the OCPP works totally different on "every" wallbox.

Perhaps i misunderstood something? Do you perhaps have any other ideas?

SKB-CGN, 2024-11-30:
Thanks a lot for your comments!

I will have a look into this again - if I remember it correctly, i set the TXProfile when the wallbox mentions, it is preparing.

Then it is accepting the profile, but not applying it.

What I don't get is, if I send the 0A DefaultTxProfile even a MeterValues request shows the 32A in return.

Will provide you with the logs after preparing the application for this scenario again!

mikuso, 2025-02-16:
Closing stale issue. If you need further help, please open a new issue. Thanks.

closed 2025-02-16 by mikuso