mikuso/ocpp-rpc #82: Remote RemoteStopTransaction rejected.

Musfick, 2024-12-08:
transactionId = 112200

the transactionId we are saving in backend

when call StartRemoteTransaction
this is our return response

```
{

status: "Accepted",

transactionId: parseInt(transactionId),

idTagInfo:{

status: "Accepted",

expiryDate: expiryDate

}

}
```

this is our RemoteStopTransaction

```
const response = await client.call('RemoteStopTransaction', {

transactionId: parseInt(transactionId), //same transaction id

});
```

but we are geeting this

0|server | { status: 'Rejected' }

0|server | Remote RemoteStopTransaction rejected.

receive meterValue but there is transactionId also StopTransaction return always transactionId 0
no. i always have unique transaction id
https://www.tktev.com/products/ev-charging-solution/ac-ev-chargers-7kw-22kw/
we are using this model

mikuso, 2024-12-09:
Hi.

This doesn't appear to be an issue with the library but more of a general OCPP query?

I'll try to help you regardless.

> when call StartRemoteTransaction this is our return response

From context, it looks like you mean `StartTransaction`, correct? Your example payload is not a valid response to `StartTransaction` because it includes an erroneous "status" field which should not be present.

> receive meterValue but there is transactionId also When unplug the charger from vehicle it give us StopTransaction and return always transactionId 0

This is most likely due to your malformed response to StartTransaction (as explained above).  If you can correct this, then the transactionId should carry through to the end of the transaction, and also then be valid for `RemoteStopTransaction`.

Hope this helps.

Musfick, 2024-12-10:
It worked after removing status from the payload
Thank you so much
Closing the issue

closed 2024-12-10 by Musfick