mikuso/ocpp-rpc #83: Help me sending operation to the chargers

rheza, 2025-01-12:
I keep getting this error, when sending operation from my web client. 

it seems the charger client is connected. but the server cant send the operation to charger. why is that ? there is no error on client about connection.

`GET /api/chargers/675f87e6ebe3d513d8413819 200 305.011 ms - 610
[2025-01-12T06:49:21.067Z] [INFO]: Executing operation RemoteStartTransaction on charger 12312
{
  "params": {
    "connectorId": 1
  }
}
[2025-01-12T06:49:21.077Z] [INFO]: Checking charger 12312 connection state: Not Connected
[2025-01-12T06:49:21.086Z] [ERROR]: Charger 12312 not connected. WebSocket state: undefined
[nodemon] files triggering change check: logs/ocpp.log
`

this is my chargers.js when the web client called, 
`const express = require('express');
const router = express.Router();
const auth = require('../middleware/auth');
const Charger = require('../models/Charger');
const ChargingSession = require('../models/ChargingSession');
const OCPPServer = require('../ocpp/OCPPServer');
const logger = require('../utils/logger');

// Get all chargers
router.get('/', auth, async (req, res) => {
    try {
        const chargers = await Charger.find();
        res.json(chargers);
    } catch (error) {
        logger.error('Error fetching chargers:', error);
        res.status(500).json({ message: 'Server error' });
    }
});

// Get charger by ID
router.get('/:id', auth, async (req, res) => {
    try {
        const charger = await Charger.findById(req.params.id);
        if (!charger) {
            return res.status(404).json({ message: 'Charger not found' });
        }
        res.json(charger);
    } catch (error) {
        logger.error('Error fetching charger:', error);
        res.status(500).json({ message: 'Server error' });
    }
});

// Handle OCPP operations
router.post('/:serialNumber/operations', auth, async (req, res) => {
    try {
        const { serialNumber } = req.params;
        const { operation, params } = req.body;

        // Get the global OCPP server instance
        const ocppServer = req.app.get('ocppServer');
        if (!ocppServer) {
            return res.status(500).json({ status: 'Rejected', message: 'OCPP server not initialized' });
        }

        // Check if charger exists and is connected
        const charger = await Charger.findOne({ serialNumber });
        if (!charger) {
            return res.status(404).json({ status: 'Rejected', message: 'Charger not found' });
        }

        const chargerConnection = ocppServer.chargers.get(serialNumber);
        
[... 742 lines cut ...]
                    await Charger.updateOne(
                        { serialNumber: chargerSerial },
                        {
                            $set: {
                                lastMeterValue: {
                                    timestamp: lastMeterValue.timestamp,
                                    value: parseFloat(energyValue.value)
                                }
                            }
                        }
                    );
                }
            }

            logger.debug(`Meter values handled for ${chargerSerial}`);
        } catch (error) {
            logger.error(`Error handling meter values for ${chargerSerial}:`, error);
            throw error;
        }
    }
}

module.exports = OCPPServer;`

Please help, basically i wanted to develop charging central system to control the charger.

rheza, 2025-01-12:
it seems there is no connection open eventhough i have succesfully connected my charger to the ocpp server.

`GET /api/chargers/675f87e6ebe3d513d8413819 200 314.002 ms - 610
[2025-01-12T07:36:06.384Z] [INFO]: Executing operation RemoteStartTransaction on charger 12312
{
  "params": {
    "connectorId": 1
  }
}
[nodemon] files triggering change check: logs/ocpp.log
[nodemon] changes after filters (before/after): 1/0
[2025-01-12T07:36:06.392Z] [ERROR]: Error executing operation: Cannot make call while socket not open
{
  "stack": "Error: Cannot make call while socket not open\n    at RPCClient._call (/root/rhechargesystem/node_modules/ocpp-rpc/lib/client.js:300:19)\n    at Queue._next (/root/rhechargesystem/node_modules/ocpp-rpc/lib/queue.js:39:35)\n    at /root/rhechargesystem/node_modules/ocpp-rpc/lib/queue.js:22:18\n    at new Promise (<anonymous>)\n    at Queue.push (/root/rhechargesystem/node_modules/ocpp-rpc/lib/queue.js:15:16)\n    at RPCClient.call (/root/rhechargesystem/node_modules/ocpp-rpc/lib/client.js:293:38)\n    at /root/rhechargesystem/routes/chargers.js:65:61\n    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)"
}
POST /api/chargers/12312/operations 500 335.634 ms - 72
GET /api/chargers/675f87e6ebe3d513d8413819 304 259.725 ms - -
[2025-01-12T07:36:12.564Z] [WARN]: Charger 12312 ping timeout
[nodemon] files triggering change check: logs/ocpp.log
`

any direction i can go though and read how to implement start and call operation correctly ?

rheza, 2025-01-12:
[image]
Error console.

rheza, 2025-01-12:
The status from the charget seems okay.
`GET /api/chargers/675f87e6ebe3d513d8413819 304 300.377 ms - -
[2025-01-12T11:16:32.574Z] [DEBUG]: [Connection] Status check for 12312:
{ 
  "bootNotificationReceived": true,
  "wsReadyState": 1,
  "clientConnected": true,
  "status": "Connected"
}
[nodemon] files triggering change check: logs/ocpp.log
[nodemon] changes after filters (before/after): 1/0

[2025-01-12T11:17:27.720Z] [WARN]: Charger 12312 WebSocket is not open for operations
POST /api/chargers/12312/operations 400 264.117 ms - 72`

but when i sending start operation. the server complained with this message, websocket not open for operations.

mikuso, 2025-01-12:
When I look at your chargers.js script, I see the following issue:

```js
// Check WebSocket connection state
const isConnected = client?.connection?.ws?.readyState === 1; // 1 = OPEN
```

This is incorrect. RPCClient does not have a connection property, so `isConnected` will therefore always be false.

There is no supported to way to directly query whether a RPCClient is currently connected or not. Instead, you can either:
- Listen for the `disconnect`, `close` and `open` events and record the state of the connection in your own code as it changes.
- Attempt to call the operation without knowing if it's connected or not. In some scenarios, the RPCClient will queue the call and send it when the connection is ready; Alternately, the call will be rejected with an error which you can handle afterwards.

closed 2025-02-16 by mikuso