mikuso/ocpp-rpc #84: Save Clent to DB

faridsya, 2025-01-20:
I am new to OCPP protocol and also at beginner level in NodeJS. How  save a client that is connected to a server into database, especially MySQL?
I got this code: let charger = Charger.findOne({identity: client.identity}); without knowing what the structure of the model is.
Thanks

mikuso, 2025-01-20:
Hi @faridsya ,

OCPP is a protocol based on WebSocket connections, so it's not possible to store an actual reference to a client in a database.

It is possible of course to store information about your chargers, especially useful if you're trying to develop an OCPP server. For instance, you may want to store authentication information, transaction history, configuration information, or the last-known state of your chargers. As for what data to store and what the structure of it looks like, I must direct you towards to OCPP specifications for all of that info - as you won't find any of it here.

You can download the specifications here: https://openchargealliance.org/protocols/open-charge-point-protocol/

faridsya, 2025-01-21:
Ok thanx for the explanation

closed 2025-01-21 by faridsya