mikuso/ocpp-rpc #86: catch TimeoutError?

zekageri, 2025-04-10:
Hi! Great library!
I'M using a charger station simulator for developing.
If I send a bad `connectorId` on `RemoteStartTransaction` the simulator does not answer.
After some delay my server crashes.

```js
const timeoutError = new TimeoutError("Call timeout");
                                     ^

TimeoutError: Call timeout
    at RPCServerClient._call (D:\projects\test\test\node_modules\ocpp-rpc\lib\client.js:361:38)
    at Queue._next (D:\projects\test\test\node_modules\ocpp-rpc\lib\queue.js:39:35)
    at D:\projects\test\test\node_modules\ocpp-rpc\lib\queue.js:22:18
    at new Promise (<anonymous>)
    at Queue.push (D:\projects\test\test\node_modules\ocpp-rpc\lib\queue.js:15:16)
    at RPCServerClient.call (D:\projects\test\test\node_modules\ocpp-rpc\lib\client.js:293:38)
    at D:\projects\test\test\dist\services\rpc\rpcOutgoing.js:39:39
    at Generator.next (<anonymous>)
    at D:\projects\test\test\dist\services\rpc\rpcOutgoing.js:8:71
    at new Promise (<anonymous>)

Node.js v22.13.1
```

Can I capture this TimeoutError somehow?

mikuso, 2025-04-11:
Aren't you able to catch this using try/catch?

e.g.
```js
try {
    await client.call('RemoteStartTransaction', { /* ... */ });
} catch (err) {
    // do something with err
    console.error(err);
}
```

zekageri, 2025-04-17:
Oh, I was looking for some handler for this case.... I'm dumb.
Will check this out! Thanks

closed 2025-05-16 by zekageri